microsoft/Qcodes #367: Example notebooks for  pacakge installed from pip.

giulioungaretti, 2016-10-25:
How to make a runnable example that people can download ?
The example notebooks , if one install as one has to a python package is impossible to find.

Downloading the notebook from the github examples won't work as the toy models are not part of qcodes pacakges.

Maybe the toy model is also a bit too hard for sombody new?  

@alexcjohnson  ideas?

alexcjohnson, 2016-10-25:
we could certainly move toymodel into the repo - maybe into instrument_drivers, since that's what it's supposed to mimic really.

But also, if folks already have *all* the examples, it would be a shame to make them download them again just because they're hard to find. What if we just made a function `qcodes.copy_examples(destination='.')`? We could put that into a little notebook so people download *one* notebook that has a little intro and then copies all the others next to it, and maybe links to some of the newly copied notebooks directly?

alexcjohnson, 2016-10-25:
I mean move toymodel into the package, of course it's already in the repo. Also you're right that it could definitely be simplified to have the same functionality without all the complicated `MockInstrument` / `MockModel` stuff. But that seems like a separate issue from making it practical for users to run the examples.

giulioungaretti, 2016-10-26:
@alexcjohnson  I think it makes more sense to just allow for the downloaded notebooks to be runnable without extra magic (just after installing qcodes). The extra function seems to me just confusing!

giulioungaretti, 2016-10-27:
The easy solution for now could be  to merge the toy things into the notebook @alexcjohnson , what you think?

alexcjohnson, 2016-10-27:
I know we're not recommending it, but that would break multiprocessing if someone wanted to try turning it on in the examples. Do you see an issue putting it in the `instrument_drivers` folder? I can do that when I add some more examples.

giulioungaretti, 2016-10-27:
@alexcjohnson ! Right,  better play safe! Let's go for instrument_divers then ! :ice_cream:

closed 2017-05-07 by giulioungaretti