microsoft/Qcodes #381: Refactor QT

giulioungaretti, 2016-11-04:
The current QtPlot is painful to work with:
  - currently impossible to save the plots
  - or to handle  qt related in any idiomatic way 
  - and much more

First step will be merging #290 ! Then polishing  all the things.

giulioungaretti, 2016-11-04:
Note to self:  this may require major changes:
- base class is too limiting
- matplotlib and qt have totally different concurrency models 
- in general using the js timer to control QT is a bit suicidal
- qt (actually everything data) shines when it's a dumb listener, not an active puller

peendebak, 2016-11-07:
@giulioungaretti @MerlinSmiles The new pyqtgraph (0.10) has been released, so merging #290 is possible. 

One issue is that qt4 is still hard-coded, e.g.
```
from PyQt4 import QtCore, QtGui 
```
I suggest replacing this with [qtpy](https://pypi.python.org/pypi/QtPy):
```
from qtpy import QtCore, QtGui
```
(als some small changes in other places)

@MerlinSmiles Can you make a new branch outside the main qcodes repo (which is read-only now). I can then make a PR for the qtpy changes.

giulioungaretti, 2016-11-07:
@eendebakpt  yep!  and yep

MerlinSmiles, 2016-11-07:
@peendebak i just had a discussion with @giulioungaretti , it seems as merging this does not work without multiproccessing (default settings now), or with severe limitations in updating the GUI :( maybe @giulioungaretti finds a fix though 😮

giulioungaretti, 2017-01-14:
Note to me and @jenshnielsen  this is more and more releavant, as we need to tackle qdev's request.
Although maybe easier to start from scratch and not use that horrible plot_object thing.