microsoft/Qcodes #396: implementing a ramp

nataliejpg, 2016-11-21:
It is possible to implement a ramp using qc.Loop sweeping a parameter and adding a wait time but I think there is some merit to discussing whether this is the way it should be done

`qc.Loop(gate.voltage.sweep(0, -10, 1)).each(qc.Wait(1)).run()`

- It creates a data set which we probably don't want
- imo its not a very natural syntax
- mostly we want to ramp for every set of a parameter and to be able to choose max ramp speed at the start of the experiment and then never worry about it again which seems like something that could go in an instrument driver and then be used in place of the plain 'set' (so no Loop at all)...

What are people's thoughts on this? It seems like something everyone would use, how should it be implemented and what would the ideal syntax be?

giulioungaretti, 2016-11-21:
Do you imagine ramp just being a set of set, every x unit time ?
If so then it could implemented in the base instruments, else it can't only implemented in the driver.

nataliejpg, 2016-11-21:
I'd want it to take whatever value ask it to set the paremeter to and use some preset minimum step (in magnitude and time) to keep setting the parameter until it is at the value you want it. I feel like @MerlinSmiles used/did this at some point but I might be imagining it?

MerlinSmiles, 2016-11-21:
I was only doing this in matlab previously...
A feature like that would be nice, also then the need for:
- async, as in many different channels ramp simultaneously
- automatic rate/step calculation so that `chA` goes from 1 to 5 during the same time as `chB` goes from 0 to 2...

giulioungaretti, 2016-11-21:
@nataliejpg  so let me rephrase, does the ramp needs just time from the outside ? or you do need to query the instrument ? 

@MerlinSmiles async, for now please forget. I mean qcodes is just not ready anything like async, there is simply no notion of concurrency ( before being confused  with the idea of using parallelism to "achieve" concurrency ).
@MerlinSmiles and yes, also interesting, but again there is no notion of channels for now

MerlinSmiles, 2016-11-21:
@giulioungaretti yeah sure, just wanted to say that before someone makes something which wont be possible to transfer at all... but will keep silent :promise:

nataliejpg, 2016-11-21:
@giulioungaretti it's needs to know what value the parameter currently has in order to calculate what to set it to next in the ramp... Whether it uses the 'last known value' or queries the instrument is up for debate.

giulioungaretti, 2016-11-21:
@nataliejpg  ok so how about this ? 
http://qcodes.github.io/Qcodes/api/generated/qcodes.StandardParameter.html 

ref:  
```
   delay (Optional[Union[int, float]]) – time (in seconds) to wait after the start of each set, whether part of a sweep or not. Can be set to 0 to go maximum speed with no errors.
max_delay (Optional[Union[int, float]]) – If > delay, we don’t emit a warning unless the time taken during a single set is greater than this, even though we aim for delay.
step (Optional[Union[int, float]]) – max increment of parameter value. Larger changes are broken into multiple steps this size.
max_val_age (Optional[Union[int, float]]) – max time (in seconds) to trust a saved value from this parameter as the starting point of a sweep.
```

giulioungaretti, 2016-11-21:
@nataliejpg  except that the docs seem to lie about max_val_age, i.e. the setting is totally disregarded.
Meaning that you can have steps but you can't have steps that **actually**  go out and ask the parameter OR use the last measured value :D Happy dining

nataliejpg, 2016-11-22:
@giulioungaretti  definitely sounds like it could work. I'll try it out. From your second comment am I right on thinking that it always uses the stored value as the value to ramp from? My other question would be which value it uses to decide how long to wait between each step of the ramp, is that delay? it's not totally clear to me from the docs. 
 I'll try it out though  :) thanks

giulioungaretti, 2016-11-22:
@nataliejpg use the source Luke, yeah docs kind of sucks. I'll sweeten this issue up with cake and better description  later on !

giulioungaretti, 2016-11-22:
@nataliejpg not the the code is any more clear but at least ain't laying: 
https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument/parameter.py#L662

giulioungaretti, 2016-11-22:
@nataliejpg  see edit above, or you'd rather have an example ?

nataliejpg, 2016-11-23:
@giulioungaretti I'll try it out today, thanks :)

nataliejpg, 2016-11-23:
@giulioungaretti 👍 works, is there, maybe we can add 2 lines to the tutorial with it in use just so that it's 'idiot proof' (or maybe we want to stop people like me who don't bother to read the docs like me from using it...) :P

giulioungaretti, 2016-11-23:
let's wait for @alexcjohnson new parameter and examples, and in case add the example there!

nataliejpg, 2016-11-23:
In that case I'll reopen as a reminder and we can close when it's done.

giulioungaretti, 2017-05-07:
@WilliamHPNielsen  I can't remember if Alex actually wrote the docs or not, probably not.
Would be terrific if you could go through the docs and if not there write a small example!

WilliamHPNielsen, 2017-05-08:
See #601, which aims to solve this issue.

PR #601 (2017-05-08): docs: Improve the tutorial
Include a description on how to make a "staircase ramp"

Fixes #396

Changes proposed in this pull request:
- Show how to make a "staircase ramp" in the tutorial
- Add more words to the tutorial and make the examples slightly less contrived

@giulioungaretti
closed 2017-05-08 by jenshnielsen