microsoft/Qcodes #397: Hard (arraylike) parameter QtPlot setpoint linearity plotting bug

nataliejpg, 2016-11-21:
### Steps to reproduce
1. Do a Loop over a hard (arraylike) parameter where you generate the setpoints and plot it with QtPlot
```
loop = qc.Loop(y1.voltage.sweep(0,-10, 0.1)).each(v1.trace)
data = loop.get_data_set()
plot = qc.QtPlot()
plot.add(data.VNA_magnitude)
_ = loop.with_bg_task(plot.update, plot.save).run()
```

data set looks like: 
```
DataSet:
   mode     = DataMode.LOCAL
   location = 'A:/TransmonExperiments/GoogleDrive/T2acquisition/Data/6QAK3_A2_qcodes/data/014'
   <Type>   | <array_id>        | <array.name> | <array.shape>
   Setpoint | yoko1_voltage_set | voltage      | (101,)
   Measured | frequency         | frequency    | (101, 401)
   Measured | VNA_magnitude     | magnitude    | (101, 401)
   Measured | VNA_phase         | phase        | (101, 401)
started at 2016-11-18 17:10:03
```
2. load the data and plot it again
```
loaded_data = qc.load_data(loc_provider.formatter.format(loc_provider.fmt,counter='004'))
plot = qc.QtPlot()
plot.add(loaded_data.VNA_magnitude)
```

### Expected behaviour
Should look the same both times and have the setpoints you would expect on the x axis

### Actual behaviour
Looks right when you live plot it (as in step 1) but gives a nonlinear spacing error when you load the data (as in step 2) in and discards the setoints but keeps the label which is now a lie! 

from 1
[image]

from 2
[image]

with this error:
```
a:\qcodes\qcodes\plots\pyqtgraph.py:325: UserWarning: nonlinear setpoint array passed to pyqtgraph. ignoring, using default scaling.
  warnings.warn('nonlinear setpoint array passed to pyqtgraph. '
```

 Note: works in matplot in terms of using the setpoint array even when loading data back (but we cant use that because of [#380](https://github.com/QCoDeS/Qcodes/issues/380))

Note2: code generating setpoints looks like: 
```
f = tuple(np.linspace(int(start), int(stop), num=npts))
self.setpoints = ((f,), (f,))
self.shapes = ((npts,), (npts,))
```

nataliejpg, 2016-11-21:
Note3: also works correctly for plotting in the notebook after taking the data (ie without live update)
```
data = qc.Loop(v1.power.sweep(-60,-10, 1)).each(v1.trace).run()
qc.QtPlot(data.VNA_magnitude)
```

nataliejpg, 2016-11-21:
Note4: this sometimes works as expected even in QtPlot (depends on the setpoints)  which makes me think its something to do with rounding in the driver producing a nonlinear setpoint array but it still wouldn't really explain (to me at least) why it would always work fine with the Note3 example but not with loaded in data.

MerlinSmiles, 2016-11-21:
I guess you could try to increase the `MAXPX` value in the `qcodes/plots/pyqtgraph.py` file to see if it is related to rounding errors?
Not that I understand how that error is actually calculated...

nataliejpg, 2016-11-23:
@MerlinSmiles that works, thanks Merlin. Not sure whether to close the issue or leave open to flag up that this is a minor thing that should be looked at/changed in qcodes. @giulioungaretti ?

MerlinSmiles, 2016-11-23:
@nataliejpg dont close it, its still a bug.

jenshnielsen, 2017-05-04:
@nataliejpg I think this has been fixed some time ago ?

nataliejpg, 2017-05-04:
@jenshnielsen agreed

closed 2017-05-04 by jenshnielsen