microsoft/Qcodes #414: error with MatPlot and qt5

peendebak, 2016-12-11:
### Steps to reproduce
1. Start python with qt5 backend 
2. Run `from qcodes.plots.qcmatplotlib import MatPlot; MatPlot()`
3. An error is generated

### Expected behaviour
No error should occur.

### Actual behaviour
The following error is generated:
```
Traceback (most recent call last):

  File "<ipython-input-1-9f230cb3ab97>", line 1, in <module>
    from qcodes.plots.qcmatplotlib import MatPlot; MatPlot()

  File "D:\users\eendebakpt\Qcodes\qcodes\plots\qcmatplotlib.py", line 43, in __init__
    self._init_plot(subplots, figsize, num=num)

  File "D:\users\eendebakpt\Qcodes\qcodes\plots\qcmatplotlib.py", line 59, in _init_plot
    figsize=figsize)

  File "d:\software\Anaconda\envs\eendebakpt\lib\site-packages\matplotlib\pyplot.py", line 1178, in subplots
    fig = figure(**fig_kw)

  File "d:\software\Anaconda\envs\eendebakpt\lib\site-packages\matplotlib\pyplot.py", line 527, in figure
    **kwargs)

  File "d:\software\Anaconda\envs\eendebakpt\lib\site-packages\matplotlib\backends\backend_qt5agg.py", line 43, in new_figure_manager
    return new_figure_manager_given_figure(num, thisFig)

  File "d:\software\Anaconda\envs\eendebakpt\lib\site-packages\matplotlib\backends\backend_qt5agg.py", line 50, in new_figure_manager_given_figure
    canvas = FigureCanvasQTAgg(figure)

  File "d:\software\Anaconda\envs\eendebakpt\lib\site-packages\matplotlib\backends\backend_qt5agg.py", line 216, in __init__
    super(FigureCanvasQTAgg, self).__init__(figure=figure)

  File "d:\software\Anaconda\envs\eendebakpt\lib\site-packages\matplotlib\backends\backend_qt5agg.py", line 65, in __init__
    super(FigureCanvasQTAggBase, self).__init__(figure=figure)

  File "d:\software\Anaconda\envs\eendebakpt\lib\site-packages\matplotlib\backends\backend_qt5.py", line 239, in __init__
    super(FigureCanvasQT, self).__init__(figure=figure)

TypeError: 'figure' is an unknown keyword argument
```
### System
Windows, qcodes master commit 525949cee77be3f188e4a9c5085805c665b624dd

Rubenknex, 2016-12-11:
I had this error some time ago with an application I wrote myself. Then it was due to not using the appropriate matplotlib backend corresponding to the Qt version, but this doesn't seem to be the problem here.

Maybe it is something internal in matplotlib, since you're only doing a simple plt.subplots call?

giulioungaretti, 2016-12-11:
Looks like something related to matplotlib ! What is the output of pip freeze ?

jenshnielsen, 2016-12-12:
This is indeed caused by mixing QT5 and QT4 internally in the matplotlib backend. 
PYQT5 switched to cooperative inheritance for subclasses.

Can you check if you have the `QT_API` environmental variable set. 

In windows that would be 

```
echo %QT_API%
```
in a cmd shell

eendebakpt, 2016-12-15:
The issue is related to Spyder using a different version of Qt then matplotlib in the ipython console. I can work around it by changing the Qt backend in the spyder preferences.

giulioungaretti, 2017-01-03:
Should we close then ?

peendebak, 2017-01-03:
@unga I'll close the issue. It is something that needs to be fixed upstream.

closed 2017-01-03 by peendebak