microsoft/Qcodes #415: SignalHound_USB_SA124B  print statement.

giulioungaretti, 2016-12-12:
@elrama- @AdriaanRol 

The print statement in this driver are pretty interesting!
Is there any reason for https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument_drivers/signal_hound/USB_SA124B.py#L271 ? 

I am thinking of replacing  the three print statement with a single log.error with a multi line string!

AdriaanRol, 2016-12-13:
@giulioungaretti I agree on using logging for these things. My preference would be logging.warning but my guess is you want to build in some logging module for QCoDeS? 

In any case @elrama- is on holidays now so I don't expect him to react, but I do remember that he spent some time on these error messages as it took some time to figure out how the driver worked. 

Good to see driver improvements happening 👍

giulioungaretti, 2016-12-13:
@AdriaanRol  https://github.com/QCoDeS/Qcodes/pull/416/files#diff-227f5982ce997835113c6c7997170267R271 is my proposal :D 
If you raise after, it should be an logging.error rather than a warning but that it's more of a decision the driver artist has to make. 

Yes you guessed right :D !

PR #1154 (2018-07-20): Driver / signalhound use standard logging
Fixes #415 and fixes #292 

Changes proposed in this pull request:
- Make the signalhound driver use standard logging like other drivers.
- Dust off a few strings.

@QCoDeS/core @AdriaanRol
closed 2018-07-20 by WilliamHPNielsen