microsoft/Qcodes #418: Conda install can fail

jenshnielsen, 2016-12-15:
To test our install instructions, I tried following them on Windows.

This failed with the rather unhelpful error 

```
PaddingError: Placeholder of length '30' too short in package qt-5.6.2-vc14_0.% The package must be rebuilt with conda-build > 2.0.
````

when installing qt as a dependency of matplotlib in the environment. This is because regardless of the fact that I just installed the latest version of anaconda `conda`and `conda-build` is out of date in the root env. Perhaps we should document updating the root env?

giulioungaretti, 2016-12-15:
Weeeird! Just tried on a bunch of win 7 computers, can't seem to reproduce though !

jenshnielsen, 2016-12-15:
Ok, it's likely to be transcient and gone with a new version of the anaconda installer, perhaps I got it at the wrong moment and they have since fixed it or perhaps there was something else broken on my computer.

For the record i fixed it by doing a `conda update --all` in the root environment after which everything was fine in the qcodes env. I have earlier seen someting similar with conda on linux (if I remember correctly it was related to them switching numpy to be build against mkl) where every new environment was broken until the root was updated

closed 2016-12-15 by jenshnielsen

ResidentMario, 2017-01-04:
I can confirm that I had the same issue, and that the `conda update --all` trick fixed it. 

Note that this bug has nothing to do with `Qcodes` in particular, but this page is the first page you get when you Google the error message.

sc28, 2017-01-13:
I also had the same error, however `conda update --all` did not work for me (same error occured).
What did solve the problem was to **first** run `conda update conda` (see e.g. [https://github.com/conda/conda/issues/4160](url)

delallea, 2017-01-19:
I'm also having this error in a virtual environment I created with Anaconda 32 bit on Windows, with Python 3.5:
```
PaddingError: Placeholder of length '30' too short in package qt-5.6.2-vc14_3.%
The package must be rebuilt with conda-build > 2.0.
```

The environment was created from a new Anaconda 32 bit install last week, and ``conda update --all`` updates nothing (and ``conda update conda`` tells me package conda is not installed)

jenshnielsen, 2017-01-20:
@delallea You need to update the conda package and possibly others in the root environment not the new environment that you have just created

giulioungaretti, 2017-01-20:
@jenshnielsen I just had again the same problem in win 10.
The reason is the conda package in the root environment is out of date as @jenshnielsen said.

The catch all solution is to switch to the default environment (aka root) and do:

```conda update --all```

delallea, 2017-01-20:
@jenshnielsen Thanks! Indeed ``conda upate conda`` in the root environment allowed me to install qt in my virtualenv.