microsoft/Qcodes #448: hdf5 formatter cannot write a dataset twice

peendebak, 2017-01-15:
When calling the `write_copy` method of the dataset on an existing location an error is generated.

A minimal example:

```
import numpy as np
import qcodes
from qcodes.tests.data_mocks import DataSet2D
from qcodes.data.hdf5_format import HDF5Format

qcodes.DataSet.default_formatter = HDF5Format()
    
dataset = DataSet2D()
dataset.write()
newlocation = dataset.location
dataset.write_copy(location=newlocation)
```
And another variation
```
from qcodes.tests.data_mocks import DataSet2D
from qcodes.data.hdf5_format import HDF5Format,HDF5FormatMetadata
qcodes.DataSet.default_formatter = HDF5Format()
    
dataset = DataSet2D()
dataset.write()
dataset.formatter.write(dataset, force_write=True)
```
@core @AdriaanRol

AdriaanRol, 2017-01-15:
@peendebak, could you copy paste the error here?

peendebak, 2017-01-15:
@AdriaanRol 
```

  File "<ipython-input-25-61c4dc1b0364>", line 1, in <module>
    runfile('/home/eendebakpt/svn/qutech/qtt/testing/test_dataset_bug.py', wdir='/home/eendebakpt/svn/qutech/qtt/testing')

  File "/usr/local/lib/python3.5/dist-packages/spyder/utils/site/sitecustomize.py", line 866, in runfile
    execfile(filename, namespace)

  File "/usr/local/lib/python3.5/dist-packages/spyder/utils/site/sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "/home/eendebakpt/svn/qutech/qtt/testing/test_dataset_bug.py", line 15, in <module>
    dataset.formatter.write(dataset, force_write=True)

  File "/home/eendebakpt/svn/qutech/Qcodes/qcodes/data/hdf5_format.py", line 166, in write
    group=arr_group)

  File "/home/eendebakpt/svn/qutech/Qcodes/qcodes/data/hdf5_format.py", line 224, in _create_dataarray_dset
    maxshape=(None, n_cols))

  File "/home/eendebakpt/.local/lib/python3.5/site-packages/h5py/_hl/group.py", line 108, in create_dataset
    self[name] = dset

  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper (/tmp/pip-at6d2npe-build/h5py/_objects.c:2684)

  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper (/tmp/pip-at6d2npe-build/h5py/_objects.c:2642)

  File "/home/eendebakpt/.local/lib/python3.5/site-packages/h5py/_hl/group.py", line 277, in __setitem__
    h5o.link(obj.id, self.id, name, lcpl=lcpl, lapl=self._lapl)

  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper (/tmp/pip-at6d2npe-build/h5py/_objects.c:2684)

  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper (/tmp/pip-at6d2npe-build/h5py/_objects.c:2642)

  File "h5py/h5o.pyx", line 202, in h5py.h5o.link (/tmp/pip-at6d2npe-build/h5py/h5o.c:3731)

RuntimeError: Unable to create link (Name already exists)

```

AdriaanRol, 2017-01-15:
@peendebak, looks like it is trying to write the same file twice. 

I guess there are several solutions. 
- Prevent this behaviour by creating a new file if this is the case 
- Catch this behaviour and overwrite/merge the existing file 
- Keep the warning and decide this is undesired behaviour.

eendebakpt, 2017-01-15:
I would say the second option is the right one, optionally with a warning.
The first one is very confusing.

Pieter

giulioungaretti, 2017-05-07:
@eendebakpt @AdriaanRol: I am a bit confused 🗡 

Is this happening because you have already the file there or just the folder ? 

If the first is true, then this is the expected behaviour no?

giulioungaretti, 2017-05-10:
related #596