microsoft/Qcodes #453: Cook books

giulioungaretti, 2017-01-17:
CookBooks should address the things more generally.
These were requested a year ago but like many other things ignored (#124).

- [ ] How to use the station
- [ ] How to use an instrument
- [ ] How to use parameters
- [ ] How to organize an experiment
- [ ] How to run a loop
- [ ] How to plot live data
- [ ] How to save the data
- [ ] How to read the data back, and plot it in the first place
- [ ] How to use logging and user-logging (for experiment logs)
- [ ] How to save metadata and logs

WilliamHPNielsen, 2017-01-17:
I'd certainly like to chip in on this.

peendebak, 2017-01-26:
@WilliamHPNielsen @unga Are the cookbooks like the examples in https://github.com/QCoDeS/Qcodes/tree/master/docs/examples?

I think if would help to add a `README.md` in the `qcodes/docs/examples` directory with a list of the most important examples (`Tutorial.ipynb` would be on top of the list). Also I think the `Tutorial.ipynb` should be simplified and some examples in that notebook moved to other examples.

giulioungaretti, 2017-01-26:
@peendebak  yes very similar but smaller, so that it's easier to digest them (or even copy paste).
Agree with the tutorial, it was never written for somebody that is really a beginner, as it has been show in qdev  💥💥