microsoft/Qcodes #465: SweepValues, Trobules ?

giulioungaretti, 2017-01-25:
Is this the behavior one wants?

```python
from unittest import TestCase
from qcodes.instrument.parameter import ManualParameter

from qcodes.utils.validators import Numbers

class TestSweepValues(TestCase):
    def setUp(self):
        self.c0 = ManualParameter('c0', vals=Numbers(-10, 10))

    def test_slicing(self):
        c0 = self.c0
        self.assertEquals(list(c0[1:2:1]), list(c0.sweep(1,2,1)))
```
Run tests

```python
>>>AssertionError: Lists differ: [1] != [1.0, 2.0]
``` 

- The types are different (it's python so meh, but hella confusing none the less)
- The sweep values one gets  are very different 

@alexcjohnson   maybe it's by design ?

damazter, 2017-01-25:
I remember that we deliberately kept the slice notation the same as the default python slice behaviour. This is probably at the root of this problem
I am personally not surprised by this behaviour. (I cannot say anything about the types though)

alexcjohnson, 2017-01-25:
@damazter right - `.sweep` was added because some people (understandably) didn't like the missing endpoint from slice notation, or wanted the ability to specify the number of points rather than the step. But slice notation is well defined in python - and never includes the endpoint - so we don't want to alter that.

We could debate whether coercing always to float or leaving ints where it's possible is better... probably doesn't have any practical effect at the moment but definitely confusing.

giulioungaretti, 2017-01-25:
Reminds me of this:
[image]
](url)

giulioungaretti, 2017-01-26:
jokes aside. I think it's really too confusing.
I will deprecate and then remove in the next version.

Background:
```python
In [8]: range(0,100)[0:100:1]
Out[8]: range(0, 100)
``` 

If one  thinks in terms of generators then slicing is also not supported.

Lastly while slicing is pythonic, it has to do with existing sequences (that's also why by default one can't really slice a generator as mentioned above).

This syntax creates a sequence by using a slice, and hence the behavior of the syntax is not predictable, unless one has been involved in the discussion. Moreover  it is impossible to document.

I am deeply sorry for those who used this syntax. 
☮