microsoft/Qcodes #467: Keithely 2600 "resolution"

giulioungaretti, 2017-01-27:
@MerlinSmiles  right now  we are limiting the set to 8 digits (https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument_drivers/tektronix/Keithley_2600.py#L23)
Afaik it can go to to 12 digits. Do you confirm ?

MerlinSmiles, 2017-01-27:
@giulioungaretti no hardware around in this train, try it out? stupid mistake if it can go to 12 digits.

giulioungaretti, 2017-01-27:
@MerlinSmiles it seems to go that low on the 2614B.

I am not 100% sure but the manual reports:

| Range  | Programming Resolution | Accuracy (1 Year), 23°C ±5°C ±(% rdg. + amps) | Typical Noise (peak-peak) 0.1Hz–10Hz |
|--------|------------------------|-----------------------------------------------|--------------------------------------|
| 100 nA | 2 pA                   | 0.06% + 100 pA                                | 5 pA                                 |

giulioungaretti, 2017-01-27:
@MerlinSmiles  

[image]

giulioungaretti, 2017-01-27:
Let me know if you agree with the way I read the manual 📄

MerlinSmiles, 2017-01-28:
@giulioungaretti looks very reasonable. Could you check if you can send 12 digits when in >1A range? And also how many digits do you get back when reading?

PR #485 (2017-02-09): fix: Keithely 2600 precision
According to the manual the keith can go that low.

Closes #467

@QCoDeS/core
commit c1278b22 (2017-02-09): fix: Keithely 2600 precision (#485)

According to the manual the keith can go that low.

Closes #467
closed 2017-02-09 by giulioungaretti