microsoft/Qcodes #469: Nonsensical warning when using val_mapping

WilliamHPNielsen, 2017-01-31:
### Steps to reproduce
1. A MWE is here:

    from qcodes.instrument.parameter import StandardParameter

    def myget():
        return 0

    def myset(x):
        return

    badparam = StandardParameter('badparam',
                                                           label='Parameter working, but throwing error',
                                                           set_cmd=myset,
                                                           get_cmd=myget,
                                                           val_mapping = {1: 0, 2: 'No'})

### Expected behaviour
No warnings should arise when defining `set_cmd`, `get_cmd`, and `val_mapping`.

### Actual behaviour
When defining a val_mapping for a StandardParameter, a nonsensical warning is raised.
Note that the `val_mapping` works as expected.

### Why do I think this happens
Looking in parameters.py, the statement `if get_parser is not None and not isinstance(get_cmd, str):` does not anticipate that `get_parser` is set if a `val_mapping` is set.

### System
**MAC OS X 10.12.1**

**master**

**f1e98e96ed6d5a05529fbb8aaab3e34555a73d28**

giulioungaretti, 2017-02-01:
maybe @alexcjohnson  can weight in ?

alexcjohnson, 2017-02-01:
You're right, that warning is not true anymore and should be removed. I believe what happened was when we first introduced `get_parser` and `val_mapping`, people were worried that allowing all of these things to be used together would be confusing and unnecessary. Somewhere along the line that restriction was removed but the warning was not.

WilliamHPNielsen, 2017-02-02:
Okay, I'll remove it.

closed 2017-02-03 by WilliamHPNielsen