microsoft/Qcodes #475: Dirver test.

giulioungaretti, 2017-02-03:
We need some solution (that should have been baked into the base class to begin with) to make sure the drivers are actually valid python.

jenshnielsen, 2017-02-03:
I.e #474 
and 
in #459 the default values could not validate

giulioungaretti, 2017-02-03:
@jenshnielsen  I am thinking about having a mock, that simply replaces base, and tests if all the instruments at least can be instantiated.

jenshnielsen, 2017-02-03:
Yes agreed, have been doing that manually when testing drivers for instruments that I don't have easy access to. Above is just some examples of why we need it

sohailc, 2017-08-16:
Making a mock instrument server which receives messages and sends reply's on specified ports on localhost is I think the best option for making being able to run and debug instrument drivers, at least for IP instruments.  In this way we can leave the driver unmodified, that is, the driver being tested is exactly the same as the driver that will actually be used. I have made mock instrument server in my PR (https://github.com/QCoDeS/Qcodes/pull/700), but maybe it is a good idea to open a seperate PR specifically for this problem.

WilliamHPNielsen, 2017-12-18:
I guess this was dealt with by #859. At least for VISA instruments there is now a standard way of doing this.