microsoft/Qcodes #477: PyQTgraph Plotting API hard to add legend

AdriaanRol, 2017-02-06:
### Expected behaviour
I want to add a legend to a plot by doing something similar to the example below 

```python 
from qcodes.plots.pyqtgraph import QtPlot
vw = QtPlot(windowTitle='My Plot') # vw stands for viewing window 
# a is some analysis object that contains my data, labels and legend
for i in range(4): 
    vw.add(x=a.sweep_points, y=a.measured_values[i], 
                xlabel=a.xlabel, ylabel='Contrast',  symbol='o', symbolSize=5, 
              name=labels[i])  
vw.subplots[0].addLegend() # this acts on the pyqtgraph object directly and is not part of the qcodes API
```
The above example does not work. Instead I have to hack my way around it and use the following snippet 

```python 
vw.clear()
vw.subplots = [vw.add_subplot()]
vw.subplots[0].addLegend()
for i in range(4): 
    vw.add(x=a.sweep_points, y=a.measured_values[i], xlabel=a.xlabel, ylabel='Contrast',  symbol='o', symbolSize=5, 
           name=labels[i])
``` 
This is undesirable because 
- it is hard to add a Legend
- I need to know the internals of the subplot object and why I need to expand the trace first (by hand). 
- the legend itself looks visually bad (gray background spaces in beginning are ignored). 
It results in the following image. 
[image]

I propose improving the plotting API by adding a simple addLegend() function. 
@nataliejpg , I believe you also had some suggestions to improve the API. Let's collect them in this issue.

nataliejpg, 2017-02-06:
Thanks @AdriaanRol, would genuinely help me a lot if someone linked a good QtGraph cookbook or similar or an example of the how to do basic things with the qcodes version. Seems like a lot of it is under the hood (like the default_parameter_array which I think it's called which by default gets plotted) and how it automatically plots against setpoints but only sometimes and how you crash QtGraph if you try to plot data with a non trivial shape and then have to restart the notebook (or sometimes the computer) to get it back which is pretty bad! I realise these things are light on the examples though so I will be better at sticking things on here when I come across them.

AdriaanRol, 2017-02-06:
@nataliejpg , not a cookbook but it should be quite useful. I use the QCoDeS pyqtgraph API to do live plotting, it shows you how to update existing plots and create new windows. I extensively use the .clear() to reuse my plotting windows. 
Hope this helps. 
https://github.com/DiCarloLab-Delft/PycQED_py3/blob/master/pycqed/measurement/measurement_control.py

MerlinSmiles, 2017-02-06:
@nataliejpg 
some examples come with pyqtgraph itself. Open the examples and find what you want to do, then check how it is done. It certainly doesnt cover all the things it can do but is a good start.

```Python
import pyqtgraph.examples as ex
ex.run()
```

giulioungaretti, 2017-02-06:
@nataliejpg @MerlinSmiles   the main issue is the the pyqtgraph api is so buried under a series of abstractions + it's so spaghettified with the data set. If you look at @AdriaanRol  code, you see that most of the things are de-spaghettfied and added to the plot.

We hope to fix the api soon <3 in the mean time  I hope to give you some examples to do what you want to do asap.

nataliejpg, 2017-02-06:
thanks @giulioungaretti @MerlinSmiles @AdriaanRol  :)

MerlinSmiles, 2017-05-18:
I guess this was fixed @AdriaanRol ?

AdriaanRol, 2017-05-18:
AFAIK this was added

giulioungaretti, 2017-05-18:
Yes it was! Closing :D

closed 2017-05-18 by giulioungaretti