microsoft/Qcodes #478: Decadac reset when restarting the kernel

fabioansaloni, 2017-02-06:
If you encounter a bug use the following template.
If you have a feature request feel free to freestyle.
### Steps to reproduce
1.  deca.ch0_voltage(1)
2.  restart the kernel
3.  deca.ch0_voltage()

### Expected behaviour
Channel 0 should still be set at 1 V after the kernel has been restarted

### Actual behaviour
Channel 0 is set to 0 V once the kernel has been restarted

### System
**operating system**
Windows 7
**qcodes version**
0.1.2

jenshnielsen, 2017-02-06:
The only thing the decadac driver seems to be doing when initialised seems to be to set mode to 1 which is needed to ensure that the decadac is active https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument_drivers/Harvard/Decadac.py#L101

I guess that this also resets all voltages but there does not seem to be a way to get the mode setting

giulioungaretti, 2017-02-07:
@WilliamHPNielsen @jenshnielsen @fabioansaloni  and afaik there is no way to query for the mode value (firmware issues). In short, either you make sure the dacadac is **actually** working but this will probably reset all the voltages or you live on the edge and your decadac may be off all the times without you noticing. This is a limitation of the firmware.

WilliamHPNielsen, 2017-02-07:
@fabioansaloni So you will be much better off using the QDac. The driver for it is online as of now!

giulioungaretti, 2017-02-08:
@fabioansaloni @jenshnielsen @WilliamHPNielsen it seems like it's a decadac bug (surprise).

WilliamHPNielsen, 2017-06-23:
With the new Decadac driver, is this solved? @fabioansaloni?

WilliamHPNielsen, 2017-06-28:
Closing this issue for now as the setup whence the issue came no longer exists.

closed 2017-06-28 by WilliamHPNielsen