microsoft/Qcodes #489: qcodes init issue

nataliejpg, 2017-02-11:
### Steps to reproduce
1.  try to import qcodes 
2. fail

### Actual behaviour
In tutorial notebook run
```
%matplotlib nbagg
import matplotlib.pyplot as plt
from pprint import pprint
import time
import numpy as np
import qcodes as qc
```
Get
```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-2-a6df1f507aa7> in <module>()
      4 import time
      5 import numpy as np
----> 6 import qcodes as qc

a:\qcodes\qcodes\__init__.py in <module>()
     42 
     43 from qcodes.station import Station
---> 44 from qcodes.loops import get_bg, halt_bg, Loop
     45 from qcodes.measure import Measure
     46 from qcodes.actions import Task, Wait, BreakIf

a:\qcodes\qcodes\loops.py in <module>()
     55 from qcodes import config
     56 from qcodes.station import Station
---> 57 from qcodes.data.data_set import new_data, DataMode
     58 from qcodes.data.data_array import DataArray
     59 from qcodes.data.manager import get_data_manager

a:\qcodes\qcodes\data\data_set.py in <module>()
     11 from .gnuplot_format import GNUPlotFormat
     12 from .io import DiskIO
---> 13 from .location import FormatLocation
     14 from qcodes.utils.helpers import DelegateAttributes, full_class, deep_update
     15 

a:\qcodes\qcodes\data\location.py in <module>()
     18 
     19 
---> 20 class FormatLocation:
     21 
     22     """

a:\qcodes\qcodes\data\location.py in FormatLocation()
     84     """
     85 
---> 86     default_fmt = config['core']['default_fmt']
     87 
     88 

a:\qcodes\qcodes\config\config.py in __getitem__(self, key)
    386     def __getitem__(self, key):
    387         if '.' not in key:
--> 388             return dict.__getitem__(self, key)
    389         myKey, restOfKey = key.split('.', 1)
    390         target = dict.__getitem__(self, myKey)

KeyError: 'default_fmt'
```

### System
Windows

**qcodes branch**
master

**qcodes commit**
 ac194581a38ca53a0d7c73a4f5a3d09fc5deb075

giulioungaretti, 2017-02-11:
@nataliejpg  most likely you have an old config file in your home or working dir 🦄 can you check if that's the case?

nataliejpg, 2017-02-11:
I think it is because it works on my mac

nataliejpg, 2017-02-11:
Best way to resolve? Delete and redo all the config? :( would be good if updates didn't break because you have previously added a config file (or if we could have some warning).

nataliejpg, 2017-02-11:
yup works after that but can we fix it or put a warning/instructions if something is not backward compatible? Think this is still an issue until that is addressed. Maybe worth renaming though

giulioungaretti, 2017-02-13:
@nataliejpg  that's a good point. I will think about this ! thanks!

closed 2018-09-21 by nataliejpg