microsoft/Qcodes #494: Standardize unit

giulioungaretti, 2017-02-14:
Go over all instrument and set to the "correct value".
Example:  mV should be V and so on.
This includes converting when an instrument gets values in mV .

It's then up to the plotting library do to the scaling.

giulioungaretti, 2017-02-14:
this also implies removing units from the label

peendebak, 2017-02-14:
@unga I am not sure I like the idea of converting units to the "correct" value. What is correct then? For me the IVVI units in mV make sense, I dont want them in V.

AdriaanRol, 2017-02-14:
About a year ago we had a pretty similar discussion on the merits of this in issue #56 . I don't remember exactly what the conclusions were but it may be worth including the lessons from that discussion here.

giulioungaretti, 2017-03-09:
I think this will make some unhappy, but it's mostly unavoidable.
To have a uniform way of doing scaling and plotting, we need to pick a standard.
And it will be si and derived units !

closed 2017-03-09 by giulioungaretti