microsoft/Qcodes #498: Multi

jenshnielsen, 2017-02-20:
If you encounter a bug use the following template.
If you have a feature request feel free to freestyle.
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### System
**operating system**

**qcodes branch**

**qcodes commit**

jenshnielsen, 2017-02-20:
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-7-3209025a3c3c> in <module>()
----> 1 data = qc.Measure(a.acquisition).run()

/Users/jhn/src/Qcodes/qcodes/measure.py in run(self, use_threads, quiet, data_manager, station, **kwargs)
     76 
     77         data_set = self._dummyLoop.get_data_set(data_manager=data_manager,
---> 78                                                 **kwargs)
     79 
     80         # set the DataSet to local for now so we don't save it, since

/Users/jhn/src/Qcodes/qcodes/loops.py in get_data_set(self, data_manager, *args, **kwargs)
    721 
    722             data_set = new_data(arrays=self.containers(), mode=data_mode,
--> 723                                 data_manager=data_manager, *args, **kwargs)
    724 
    725             self.data_set = data_set

/Users/jhn/src/Qcodes/qcodes/loops.py in containers(self)
    521                 # note that this supports lists (separate output arrays)
    522                 # and arrays (nested in one/each output array) of return values
--> 523                 action_arrays = self._parameter_arrays(action)
    524 
    525             else:

/Users/jhn/src/Qcodes/qcodes/loops.py in _parameter_arrays(self, action)
    594             for j, (vij, nij, lij) in enumerate(zip(sp_vi, sp_ni, sp_li)):
    595                 sp_def = (shape[: 1 + j], j, setpoints, vij, nij, lij)
--> 596                 if sp_def not in all_setpoints:
    597                     all_setpoints[sp_def] = self._make_setpoint_array(*sp_def)
    598                     out.append(all_setpoints[sp_def])

TypeError: unhashable type: 'numpy.ndarray'
```

jenshnielsen, 2017-02-20:
More of an Array parameter issue probably but anyway. 

If you do a measurement of an array parameter the setpoint array is correctly created as a data array in the dataset but not labled as is_setpoint=True

paul-lehmann-unibas, 2024-02-19:
I wanted to ask whether this issue is still actively pursued? I think I also read in some places that the MultiParameter might be deprecated at some point?
I have frequent use cases for the MultiParameter, but it can be cumbersome to use.
A feature I would like the MultiParameter to have would be to support multi-dimensional non-gridded setpoints.

jenshnielsen, 2024-02-20:
This is not something that anyone to my knowledge are actively working on at the moment. None the less I think the MultiParameter sill has issues. If you have some concrete suggestions for improving the MultiParameter (or an alternative design) we are definitely happy to discuss that