microsoft/Qcodes #499: Units and is_setpoint is not propagated by `_parameter_arrays`

jenshnielsen, 2017-02-20:
For instance for a simple multiparameter with one array 

```

Out[8]:
{'__class__': 'qcodes.data.data_array.DataArray',
 'action_indices': (0, 0),
 'array_id': 'MyMockAlazar_rawname',
 'instrument': '__main__.MockAlazar',
 'instrument_name': 'MyMockAlazar',
 'is_setpoint': False,
 'label': 'rawlabel',
 'labels': ('rawlabel',),
 'name': 'rawname',
 'names': ('rawname',),
 'setpoint_labels': (('timelabel',),),
 'setpoint_names': (('timename',),),
 'shape': (629,),
 'unit': None,
 'units': ('v',)}
```

and

```
{'__class__': 'qcodes.data.data_array.DataArray',
 'action_indices': (0,),
 'array_id': 'timename',
 'is_setpoint': False,
 'label': 'timelabel',
 'name': 'timename',
 'shape': (629,),
 'unit': None}
```

Fix coming up in a few minuts

giulioungaretti, 2017-03-07:
this will be closed by #512, right ?

jenshnielsen, 2017-03-07:
Yes

closed 2017-03-07 by giulioungaretti