microsoft/Qcodes #513: IVVI communication error does not raise correct error

AdriaanRol, 2017-03-07:
When using the IVVI driver, if you repeatedly get a communication error it should raise that exception. 

However, because of the line https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument_drivers/QuTech/IVVI.py#L291 , the actual error is not raised. Instead it will give a very nondescript ``` File "d:\githubrepos\qcodes\qcodes\instrument_drivers\QuTech\IVVI.py", line 291, in _get_dacs
    raise('IVVI Communication error')
TypeError: ('exceptions must derive from BaseException', 'getting IVVI_dac16')```

Easy to fix by raising the original exception instead of a custom one, I think this both solves the bug and would be the intended behaviour.

giulioungaretti, 2017-03-07:
@AdriaanRol yes, indeed that is just not valid python code :D

WilliamHPNielsen, 2017-12-15:
This was fixed by #541

closed 2017-12-15 by WilliamHPNielsen