microsoft/Qcodes #517: default Instrument get_idn returns a warning

peendebak, 2017-03-09:
For any `Instrument` that does not have a `self.ask('*IDN?')` a warning is generated. This is inconvenient for all derived classes that do not have a VISA interface.

The warning is generated here:

https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument/base.py#L130

A minimal example:
```
from qcodes import Instrument

class dummy(Instrument):
    pass

d=dummy(name='test')
idn=d.get_idn()
```
Can we remove the warning (or change it into a `logging.info`)?

@giulioungaretti

AdriaanRol, 2017-03-09:
I agree with this. I think that the connect message (that raises this warning)  can be very useful. 
I would propose that we solve this problem by implementing a default `.get_idn()` that returns `self.name` (in the format that works with the message).

giulioungaretti, 2017-03-09:
@peendebak @AdriaanRol, yes agree. I guess this was written with the assumption that **all** instruments have the arcane visa IDN command.

Not sure what's best:
  - 1 move it to VISA base class
  - 2 change to info (but it's kind of meant to be a warning ?) 
  - 3 change to info, but  return a sane default instead

I would go for 1, but we have to check how many instruments inherit from IP and speak visa !

AdriaanRol, 2017-03-09:
@giulioungaretti , Another quite common use case for us is to have virtual or meta instruments. These talk to lower level instruments and generally do not have an IDN command (as they only exist in python). I think a sane default would make most sense. 

That would still be consistent with -1 but would also require a sane connect message.

peendebak, 2017-03-09:
@giulioungaretti @AdriaanRol Shall I create a PR with option 3). A default that we use right now is:
```
        IDN = {'vendor': None, 'model': self.name,
               'serial': None, 'firmware': None}
```

giulioungaretti, 2017-03-09:
@peendebak that would be really awesome!

PR #518 (2017-03-10): remove warning for get_idn, add default value
Fixes #517 .

@unga @AdriaanRol
closed 2017-03-10 by giulioungaretti