microsoft/Qcodes #528: Deleting or not ?

giulioungaretti, 2017-03-16:
Not sure it's by design (@alexcjohnson ?):

This will fail :
(does not fail in the tests somehow) 

``` python
from qcodes.tests.instrument_mocks import DummyInstrument
a = DummyInstrument()
del a
b = DummyInstrument()
``` 

This does not fail:
``` python
import gc
from qcodes.tests.instrument_mocks import DummyInstrument
a = DummyInstrument()
del a
gc.collect()
b = DummyInstrument()
```

alexcjohnson, 2017-03-16:
If `gc.collect()` has an effect, then this certainly isn't by design...
The error is:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/alex/qdev/Qcodes/qcodes/instrument/metaclass.py", line 43, in __call__
    cls.record_instance(instrument)
  File "/Users/alex/qdev/Qcodes/qcodes/instrument/base.py", line 227, in record_instance
    raise KeyError('Another instrument has the name: {}'.format(name))
KeyError: 'Another instrument has the name: dummy'
```

Which is definitely something we want to fail on... the question is
a) do we want `del a` to result in the complete removal of the instrument, or should it still be present in `DummyInstrument.instances()` until you call `a.close()`?
b) why does it matter if you explicitly collect garbage right now? I guess there's still a circular ref somewhere?

giulioungaretti, 2017-03-17:
@alexcjohnson  I stumbled upon this while tweaking this test (which does not fail, but it should :D) 
:
https://github.com/QCoDeS/Qcodes/blob/master/qcodes/tests/test_instrument.py#L960

jenshnielsen, 2017-05-31:
I stumbled over this again when working on the instrument tests (simplified):

````python
class TestInstrument(TestCase):

    def setUp(self):
        self.instrument = DummyInstrument(
            name='testdummy', gates=['dac1', 'dac2', 'dac3'])
        self.instrument2 = MockParabola("parabola")

    def tearDown(self):
        # force gc run
        del self.instrument
        del self.instrument2
        gc.collect()

    def test_something(self):
          pass

    def test_something_else(self):
        assert 1 == 2

```
Assuming that `test_something_else` runs before test_something both will fail because the instrument is not closed probably in `tearDown` and a new one cannot be created in the second run of `setUp`. 

If the first test passed successfully `__del__` is called as expected. I suspect this happens because unittest keeps a reference to `self.instrument(2)` around when the test fails. 

Some debugging confirms that `__del__` is not actually called until all the tests have run if the test happens to fail. 

For reference the behaviour of  __del__ is documented in https://docs.python.org/3/reference/datamodel.html#object.__del__
`del a` will merely decrease the ref counter to `a` and `a.__del__` will be invoked when ever the gc runs so it seems likely that there is still a reference cycle involving `a`