microsoft/Qcodes #537: AWG 5014 ref clock bad parameter name /confusion

AdriaanRol, 2017-03-23:
The AWG 5014 driver has two distinct "parameters" in the hardware, 
- Clock source -> int or ext 
- Reference oscillator -> int or ext
[image]

The recently updated 5014 driver introduces a 'parameter' ref_clock_source. 
https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument_drivers/tektronix/AWG5014.py#L160
This refers to using an internal or external clock and and has nothing to do with the reference oscillator, making this a confusing name. 
Additionally there is no parameter that sets the reference oscillator. 

The driver does contain some reference to these parameter under their proper name. 
https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument_drivers/tektronix/AWG5014.py#L847-L852

I would suggest the following fix. 
- Rename the `ref_clock_source` parameter to `clock_source`. 
- Add a parameter for the `reference_source`
 Both of these names are the names that the AWG uses internally, both in it's GUI as in it's sequencer config. 

@WilliamHPNielsen as author of the recent fix, do you agree with these changes? If so I will be sending a PR shortly.

nataliejpg, 2017-03-24:
totally agree with @AdriaanRol, have been meaning to flag this up but got distracted/waylaid. At the moment initialising the qcodes instrument seems to set the reference source to internal even if it the instrument previously was set to external but I hadn't dug into what the parameter was actually doing.

AdriaanRol, 2017-03-25:
@nataliejpg have you seen #538 ?

PR #538 (2017-03-28): Awg enhancements
Fixes #537 

Changes proposed in this pull request:
- Added `clock_source` and `ref_source` parameters , removed ref 
- Changed get parser for filter function to use INF instead of 9.9e37 (which causes a rare bug)

A motivation for the first change is in #537 
A motivation for the second change is that INF is more readable (which is nice), it is the same value under the hood, and the value 9.9e37 caused me trouble when sending it to the monitor. I suspect it will prevent more trouble in the future. Additionally, the manual of the AWG says that this INF setting corresponds to o [... cut]
commit 7ddd6b9b (2017-03-28): Chanes to close #537

Adds clock_source and ref_source parameters
Also added a change to the get parser of the lowpass filter setting. The
value 9.9e37 corresponds to infinity but can give problems when sending
it over a port (such as to pyqtgraph).
closed 2017-03-28 by WilliamHPNielsen