microsoft/Qcodes #543: Sweeping magnet in spherical coordinates can fail

MerlinSmiles, 2017-03-27:
I have an issue with a 3-axis magnet (a driver not online yet, but basically using the same spherical to cartesian conversion as the Mercury IPS)

When i sweep a value in the spherical coordinates, lets say the radius, the driver will use \theta and \phi values from cache, and calculate the xyz field. That works fine, unless the t and p values are updated. That happens i.e. when the user actually reads the spherical coordinates, if a `magnet.rtp.get()` is called at every sweep point, the internal r,t,p values will change eventually (right away, or after going through zero, or...).

The issue is basically that we have many representations of one xyz point in the spherical system. And I have no idea how to ensure that the other two variables will not change during a given sweep.

The only way to make this not happen (that i can come up with) would be to just not ask for spherical values during a sweep.

Anyone with any ideas?
@core @AdriaanRol @peendebak <- just tagging because of some reason...

jenshnielsen, 2017-03-27:
I guess we need to run the theta,phi coordinates throung a normaliser i.e. from whatever the instrument returns to theta=[0, pi], phi=[0, 2*pi] before updating the saved values? I guess we also have to somehow handle the poles at theta=0 specially as phi is essentially undefined here.

MerlinSmiles, 2017-03-27:
yes, I do have those limitations in place, but the conversions make the radius always positive, and just from the values of xyz we dont know if it is expected to be negative.

giulioungaretti, 2017-04-11:
@MerlinSmiles what/where  is the cache ?

MerlinSmiles, 2017-04-11:
@giulioungaretti 
that sweeping the radius makes a wrong sweep if you ask for the field, as it will update the internal ange values, and flip around the sphere, look at Bz here:
[image]
This is how it should look like:
[image]

giulioungaretti, 2017-04-12:
@MerlinSmiles internal meaning the magnet controller ? If so, yes the only way is to forbid asking the field.

MerlinSmiles, 2017-04-12:
Yes, but we cant really forbid that, no?

giulioungaretti, 2017-04-12:
no :D but we can try !

MerlinSmiles, 2017-04-12:
Not sure that we need to, maybe just increase awareness of that situation...

PR #897 (2018-08-03): Driver/Mercury iPS with VISA
Add a VISA driver for the MercuryiPS using the `FieldVector` setpoint technology already in use in our AMI430 driver. 

Fixes #543

Changes proposed in this pull request:
- Rewrite the MercuryiPS driver to use VISA
- Make the driver use the FieldVector to deal with singularities

Work items:

- [x] Add all relevant parameters
- [x] Use `FieldVector`
- [x] (once #859 is merged) Add a simulated instrument and a thorough test

@jenshnielsen
closed 2018-08-03 by WilliamHPNielsen