microsoft/Qcodes #546: bug: decadac getting and setting voltages fails unpredictably

nataliejpg, 2017-03-28:
### Steps to reproduce
1. start Harvard Decadac and  import a qcodes instrument instrument:

```
from qcodes.instrument_drivers.Harvard.Decadac import Decadac
sl0 = Decadac('dec_slot_0', port=5, slot=0)
```

2. run 

`sl0.ch1_voltage(0)`

3. fail as below, seems like its something to do with the mode (which is set in the init but somehow that isn't reliable), It is also not super reliable to fix by setting the mode to 1 but seems to work if I set it to 0 and back to 1 a few times...

### Expected behaviour
should set the voltage

### Actual behaviour
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-18-e543db036233> in <module>()
----> 1 sl0.ch1_voltage(0)

a:\qcodes\qcodes\instrument\parameter.py in __call__(self, *args)
    133         else:
    134             if self.has_set:
--> 135                 self.set(*args)
    136             else:
    137                 raise NotImplementedError('no set cmd found in' +

a:\qcodes\qcodes\instrument\parameter.py in _validate_and_sweep(self, value)
    945             e.args = e.args + (
    946                 'setting {} to {}'.format(self.full_name, repr(value)),)
--> 947             raise e
    948 
    949     def set_step(self, step, max_val_age=None):

a:\qcodes\qcodes\instrument\parameter.py in _validate_and_sweep(self, value)
    929             step_clock = time.perf_counter()
    930 
--> 931             for step_val in self._sweep_steps(value):
    932                 self._set(step_val)
    933                 self._save_val(step_val)

a:\qcodes\qcodes\instrument\parameter.py in _sweep_steps(self, value)
    891         state = self._latest()
    892         if state['ts'] is None or state['ts'] < oldest_ok_val:
--> 893             start_value = self.get()
    894         else:
    895             start_value = state['value']

a:\qcodes\qcodes\instrument\parameter.py in get(self)
    826         except Exception as e:
    827             e.args = e.args + ('getting {}'.format(self.full_name),)
--> 828             raise e
    829 
    830     def _valmapping_get_parser(self, val):

a:\qcodes\qcodes\instrument\parameter.py in get(self)
    821     def get(self):
    822         try:
--> 823             value = self._get()
    824             self._save_val(value)
    825             return value

a:\qcodes\qcodes\utils\command.py in __call__(self, *args)
    175             raise TypeError(
    176                 'command takes exactly {} args'.format(self.arg_count))
--> 177         return self.exec_function(*args)

a:\qcodes\qcodes\instrument_drivers\Harvard\Decadac.py in _getvoltage(self, channel)
    129         response = temp[::-1]
    130 
--> 131         rawvoltage = self._code2voltage(response, channel)
    132         actualvoltage = rawvoltage - self._offsets[channel]
    133 

a:\qcodes\qcodes\instrument_drivers\Harvard\Decadac.py in _code2voltage(self, code, channel)
    232         """
    233 
--> 234         code = float(code)
    235         translationdict = {1: lambda x: (x+1)*20/2**16-10,
    236                            2: lambda x: (x+1)*10/2**16,

ValueError: ("could not convert string to float: '0! M 0'", 'getting dec_slot_0_ch1_voltage', 'setting dec_slot_0_ch1_voltage to 0')
```

MerlinSmiles, 2017-03-29:
@nataliejpg that string you get looks like a response from the decadac upon setting the mode.
Looking into the driver, no response is read when setting the mode.

You could try setting the mode, and then calling
`dac.visa_handle.read()`
maybe a few times...

nataliejpg, 2017-03-29:
thanks @MerlinSmiles , I'll put that in the my initialisation wrapper and see if it helps, perhaps a fix in the driver where it does that after  setting the mode (as part of set_cmd) would help. Also I found another bug which is that if you initialise the instrument and then ask what the voltages are they do not correspond to what you last set them to which is maybe because the get isn't actually querying the instrument (I don't know) but at least a warning on this would be good, do you think this is the case @WilliamHPNielsen? I'm right now in the middle of measuring so am disinclined to disconnect and check what they are actually outputting!

nataliejpg, 2017-03-29:
ok tried that and answer is to run `visa_handle.read()` twice (but no more because it times out if you try more times than it has messages)

nataliejpg, 2017-03-29:
actually the above is only true sometimes. Also I now cannot get the instrument to output anything even with turning mode to 1 a lot of times. @MerlinSmiles are you (or anyone else?) actually using this instrument and driver reliably?

MerlinSmiles, 2017-03-29:
@WilliamHPNielsen this driver is putting the decadac in mode 1, which is a high resolution mode, and effectively disables output on channels 3,4,7,8,... aaaand it is not handling that extra resolution?
There is a manual on the qdev wiki, and those matlab drivers for copying functionality...

WilliamHPNielsen, 2017-03-30:
Sorry for that shitshow. Unfortunately, I don't have a Decadac at hand, but from @MerlinSmiles' helpful comments, I hope that #552 does the trick.

PR #552 (2017-05-08): fix: Respond to issue #546
Unfortunately not tested.

Should fix #546

@jenshnielsen @nataliejpg @MerlinSmiles
commit b8c0065f (2017-05-08): fix: Respond to issue #546 (#552)

* fix: Respond to issue #546

Unfortunately not tested.

Should fix #546

* fix: Remove fine mode

Remove the high res mode
closed 2017-05-08 by giulioungaretti