microsoft/Qcodes #563: Decadac doesn't get correct voltages on import

nataliejpg, 2017-04-07:
### Steps to reproduce
1. Import Harvard decadac slot
2. Do get call on one of the channels

### Expected behaviour
should return the voltage that channel is at. 

### Actual behaviour
Returns 0.03... (Consistently the same number every time on all channels which is totally unrelated to actual output, would write whole number but am on my phone so will add it in comments when I get a chance).

Any ideas? Possibly it's converting a message and there is a missing read() somewhere. I didn't get the chance to play around but if anyone has a chance to take a quick look (or while testing #552 pr). Cheers. 

@WilliamHPNielsen

spauka, 2017-04-11:
There is an implementation of a channelized version of the DecaDAC that solves this issue that you can see at spauka/Qcodes@76af2a377c852af7ef90ea8fca42d9f26e3e10bb that solves this issue.

You may want to backport the code that reads the DAC memory depending on your needs.

nataliejpg, 2017-04-15:
Thanks @spauka I'll check it out. It would be fab to have some sort of channels implementation.

WilliamHPNielsen, 2017-06-23:
Now that @spauka's Decadac driver got merged in, can we close this? @nataliejpg, are you happy?

nataliejpg, 2017-06-26:
@WilliamHPNielsen yes and also excited to try channels ;P

WilliamHPNielsen, 2017-06-26:
@nataliejpg Oh, they're fantastic.

closed 2017-06-26 by WilliamHPNielsen