microsoft/Qcodes #564: visa_handle.clear on instantiation causes some instruments to freeze

spauka, 2017-04-10:
The use of visa_handle.clear on the instantiation of a VISA instrument causes some instruments to lock up. In particular I've noticed for SERIAL instruments, this command spews some garbage on the serial port that requires the instrument to be reset before it is possible to use it again.

### Steps to reproduce
1. Plug in a buggy instrument and instantiate an instance of it.

### Actual behaviour
The instrument freezes

### System
Win 7
Using the Harvard DecaDAC
2b6d72b

giulioungaretti, 2017-04-10:
@WilliamHPNielsen ever noticed the same ?

spauka, 2017-04-11:
To expand on this issue, some versions of the VISA library automatically send `*CLS;` on receiving a viClear. This sequence seems to crash older versions of the decaDac. See: http://digital.ni.com/public.nsf/allkb/4A8D61E759174E7F86256E20005A8ABC

A couple of options we could choose between:
 - Don't viClear when opening an instrument. This is probably the simplest option, my preferred.
 - If we are worried about junk in the IO buffers, use viFlush instead:
```py3
self.visa_handle.flush(visa.constants.VI_READ_BUF_DISCARD | visa.constants.VI_WRITE_BUF_DISCARD)
```
 - Set the VISA flags given in the linked document such that we don't assume that VISA instruments support SCPI commands.

giulioungaretti, 2017-04-25:
@spauka I am not sure I understand the issue, or I guess there's two part:
-  the ni-visa library  (before version 3) sends automatically a *CLS; which some instruments don't support
- wether or not we want to have qcodes to send.clear at the beginning (whether or not this is a NoOp I couln't find the visa documentation )

spauka, 2017-05-30:
@giulioungaretti For each of your two points above:
1) Yes, this locks up our version of the harvard dac on certain computers
2) visa_handle.clear empties the IO buffers for instruments, so it isn't quite a no-op. Hence I think we do actually want to send this for instruments that support it.

I committed a new change that maintains the existing behavior of clearing the IO buffers for other instrument types, but uses viFlush for serial ports, which is a specific method for clearing buffers that only seems to exist for Serial instruments.

PR #565 (2017-05-30): fix: #564 visa_handle.clear causes some instruments to freeze
Removes visa_handle.clear when instrument is opened. This does nothing anyway as buffers for newly opened instruments

Fixes #564.

Changes proposed in this pull request:
- Remove visa_handle.clear from visa instrument instantiation.

@giulioungaretti
commit 05c5aa35 (2017-05-30): fix: #564 visa_handle.clear causes some instruments to freeze (#565)

* fix: #564 visa_handle.clear causes some instruments to freeze

removes visa_handle.clear when instrument is opened. This does
nothing anyway as buffers for newly opened instruments

* fix: #564 visa_handle.clear handle for serial instruments

Clear buffers for serial instruments using the flush method which is
only available for serial instruments, but does not perform additional
actions that may lock up the serial port.
This leaves the clear command in place for other instrument types that
do not support flush.
commit 9cbb774f (2017-05-30): Generated gh-pages for commit 05c5aa3
Author: Sebastian Pauka <spauka@gmail.com>

    fix: #564 visa_handle.clear causes some instruments to freeze (#565)
closed 2017-05-30 by giulioungaretti