microsoft/Qcodes #596: Dataset write does not write if file was written and then deleted.

giulioungaretti, 2017-05-04:
Step to reproduce:

- do a loop 
- finish the loop
- delete the file on disk 

Then try to save agin:
- data.write() 

The file won't re appear on disk.

@alexcjohnson  is there a design idea behind this behaviour that I can't see ?

@everynodyelse Is this the wanted behaviour ?

Do we want a way  to force save anyway ?

MerlinSmiles, 2017-05-05:
This has now happened several times for us when using a combining parameter, to set the same value on two gates, and sweeping that one.

giulioungaretti, 2017-05-05:
@MerlinSmiles this is reproducible also by manually messing with  ```last_saved_index``` or ``` save_range```

alexcjohnson, 2017-05-08:
@giulioungaretti no, not intentional. I didn't want to do a full check of the file to see if *all* the data is in it on every write, that would take forever. But checking whether the file exists period seems like a smart thing to do!