microsoft/Qcodes #598: Handling interrupts

giulioungaretti, 2017-05-07:
Hi all (esp you @eendebakpt @peendebak @MerlinSmiles ),
   let's use this issue to fix the problem of handling interruptions.

There are two parts to this problem:

-  How to deal with blocking instruments call:

   1.  Let the instrument finish (at all costs) unless a hard stop is requests (like  force interrupt)
   2. Break the blocking call, leaving the instrument in a potentially undefined state (if something was request but not read f.ex. of if an command write was interrupted before the entire command was written)

-  Where to stop in a loop (but since may people have their own loop that's maybe less important):
   
   1.  It does not matter 
   2.  At every sate change ( basically any I/O: send command to instrument, write to disk etc)
   3. None of the above

MerlinSmiles, 2017-05-08:
@giulioungaretti 

Here is my wishlist

- How to deal with blocking instruments call:
  1. Let every hardware call finish first, i.e. if a get or set called, let that finish.
  2. Include a timeout that would break the thing after i.e. 30 seconds (I understand that this is harder to implement, but its a wish 👼 )
  3. Stop any software ramp, i.e. dont finish setting the magnet to 5T.

- Where to stop in a loop (but since may people have their own loop that's maybe less important):
  1. just finish everything it is currently doing and then stop

- What to do after interrupt:
  1. Save Data with metadata that includes the interrupt stop

jenshnielsen, 2017-05-08:
@MerlinSmiles how should we handle an ongoing hardware sweep. Something like.

* set magnet to sweep from 0 to 5 T at a rate of 0.1 T/min
* start sweep
* measure every 60 seconds (51 points)
* interrupt after 20? points

I think most users would expect the magnet to stop at/around 2 T in that case.
But obviously that would require communication with the magnet to tell it to stop ramping when handling the keyboard interrupt.

MerlinSmiles, 2017-05-08:
@jenshnielsen 
Yes I agree, I would expect it to stop there.
Which is why i have a little hard time with the KeyboardInterrupt way of stopping the thing. A stop should gracefully stop everything, also if that takes some time.
I dont stop to get back to a kernel I can use for something ASAP, but to stop whatever measurement I was doing.
In your example, in my opinion the driver should send a stop signal to the magnet.

giulioungaretti, 2017-05-08:
I was thinking to introduce this for any parameter (being gettable or settable does not matter).

```python
import signal
import logging

class DeferInterrupt(object):
        
    def __enter__(self):
        self.counter = count()
        self.signal_received = None
        self._handler = signal.getsignal(signal.SIGINT)
        signal.signal(signal.SIGINT, self.handler)

    def handler(self, sig, frame):
        """
        Handler signint either by doing nothing
        or if got twice in a row by re-signaling
        """
        count = next(self.counter)
        self.signal_received = (sig, frame)
        if count  == 0:
            logging.debug('SIGINT received. Delaying KeyboardInterrupt.')
        if count == 1:
            logging.debug('SIGINT received Twice. Forcing exit.')
            temp_signal = self.signal_received
            self.signal_received = None
            self._handler(*temp_signal)

    def __exit__(self, type, value, traceback):
        # restore old handler
        signal.signal(signal.SIGINT, self._handler)
        if self.signal_received:
            self._handler(*self.signal_received)
```

Meaning that a single interrupt will wait for whatever to finish, double interrupt will kill forcibly!
This is for example if your set or get takes like 10 hours!

jenshnielsen, 2017-05-15:
Another issue brought up by the t10 users is that if you do something like in a cell and interrupt it doing the first loop it will continue to do the second one i.e.: 

```
do1d(dac1.dac1, 0, 10, 10, 1, dac1.dac1) # interrupt this

do1d(dac1.dac1, 10, 0, 10, 1, dac1.dac1) # this will continue running
```

This is because do1d does 
```python
try:
    _ = loop.with_bg_task(plot.update).run()
except KeyboardInterrupt:
    print("Measurement Interrupted")
```
and swallows the KeyboardInterrupt