microsoft/Qcodes #620: Data truncated when saved to disk

MerlinSmiles, 2017-05-18:
### Steps to reproduce
1. Measure a parameter with a high resolution (lots of decimals)

### Expected behaviour
Data should be measured and saved with full precision

### Actual behaviour
Data is truncated to 6 decimal places (at least in gnuplot formatter)
You get angry when a time measurement does a staircase thing.

### System
**operating system**
Win 10
**qcodes branch**
Master

@core

jenshnielsen, 2017-05-18:
Yes, the default gnuplot formatter is "%g" which has 6 digits. I am not sure thats the best. 

You should be able to work around this by creating a GnuPlotFormatter and passing 
a different number_format to it such as `.10g` to get 10 digits after the dot. I am not sure what the default should be be but I suspect we should change it.

See https://github.com/QCoDeS/Qcodes/blob/master/qcodes/data/gnuplot_format.py#L30

```python
In [11]: "%g" %  (numpy.pi)
Out[11]: '3.14159'

In [17]: "%.10g" %  (numpy.pi)
Out[17]: '3.141592654'
```

giulioungaretti, 2017-05-18:
@MerlinSmiles sorry to hear that. This is indeed embarrassingly bad. 
It happens here: https://github.com/QCoDeS/Qcodes/blob/master/qcodes/data/gnuplot_format.py#L70 .

And I guess the fact that by default it rounds six digits was either not known by the original writer or casually not-documented.

MerlinSmiles, 2017-05-18:
Yes, I saw that it was happening there, but kind of get unhappy with such a feature :(

jenshnielsen, 2017-05-18:
I don't think we disagree about that @MerlinSmiles

MerlinSmiles, 2017-05-18:
I would never doubt on that, just reporting to get attention to it :)

WilliamHPNielsen, 2018-07-18:
Indeed, #783 fixed this issue.

closed 2018-07-18 by WilliamHPNielsen