microsoft/Qcodes #621: Data mingled across columns in dataset

MerlinSmiles, 2017-05-18:
### Steps to reproduce
1. Measure something with `.run(use_threads=True)`

### Expected behaviour
Get a neat dataset

### Actual behaviour
Data mingles across columns in dataset 💘 

### System
**operating system**
Win 10

**qcodes branch**
Master

Two identical measurements:
[image]

MerlinSmiles, 2017-05-18:
just an example data from a different dataset, see the lines marked with love, columns 2 and 3 switch around...

```
-5.994    4.27718    0.0173826    0.0467188
-5.991    4.27718    0.0173826    0.0467227
-5.988    0.0173826    4.27718    0.0467227 # <3
-5.985    4.27718    0.0173826    0.0467227
-5.982    4.27718    0.0173826    0.0467227
-5.979    4.27718    0.0173826    0.0467188
-5.976    4.27718    0.0173826    0.0467227
-5.973    4.27718    0.0173826    0.0467227
-5.97    0.0173826    4.27718    0.0467227 # <3
-5.967    4.27718    0.0173826    0.0467227
-5.964    0.0173826    4.27718    0.0467227
-5.961    4.27718    0.0173826    0.0467265
-5.958    4.27718    0.0173826    0.0467265
-5.955    4.27718    0.0173826    0.0467265
-5.952    4.27718    0.0192095    0.0467303
-5.949    4.27718    0.0192095    0.0467265
-5.946    4.27718    0.0192095    0.0467265
-5.943    4.27718    0.0192095    0.0467265
-5.94    0.0192095    4.27718    0.0467265 # <3
-5.937    4.27718    0.0192095    0.0467303
-5.934    4.27718    0.0192095    0.0467303
-5.931    4.27718    0.0192095    0.0467303
```

jenshnielsen, 2017-05-18:
Thats really bad. I tried reproducing it with a couple of dummy instruments but didn't see it. Are you measuring the 3 parameters from the same instrument? I am wondering if it's the driver which is not thread safe?

giulioungaretti, 2017-05-18:
@jenshnielsen  I think that's def a possibility.

MerlinSmiles, 2017-05-18:
yup, just commented to @giulioungaretti on slack, it happens when I use two parameters on the same instrument, in this case its an SR830 lock-in

MerlinSmiles, 2017-05-18:
and works fine with only on parameter per instrument

PR #771 (2017-09-29): fix: protect threading users against clutter
Fixes #621, to some extent

Changes proposed in this pull request:
- Raise an exception if users try to multi-thread read several values from one instrument
- Add a simple test for the exception

@jenshnielsen
closed 2017-09-29 by jenshnielsen