microsoft/Qcodes #624: notebook // matplotlib confusion

giulioungaretti, 2017-05-24:
The function in_notebook  is pretty useless, and it  prevents MatPlot to be in the qcodes name space, for no good reasons.
We do not really depend on the notebook anymore (thankfully).

peendebak, 2017-05-24:
@giulioungaretti Removing the notebook check fixes the issue, but it adds overhead in importing qcodes. For people wanting to quickly import qcodes (this will probably become important again when doing multiprocessing) the overhead of importing pyqtgraph and matplotlib is significant.

giulioungaretti, 2017-05-24:
@eendebakpt @peendebak  then we can put a config switch ? see pr #625

peendebak, 2017-05-25:
@giulioungaretti The config option in #625 looks good to me.

astafan8, 2018-06-14:
I close it since the fix has been implemented in #625 and #699

closed 2018-06-14 by astafan8