microsoft/Qcodes #627: Not able to use array or list as setpoints

nulinspiratie, 2017-05-29:
When using a parameter that has custom setpoints, the loop in `ActiveLoop._parameter_arrays` forces the setpoints to be hashable, otherwise an error is raised. 
This means that setpoints as a tuple are allowed, but lists and numpy arrays are not.
This has a consequence that setpoints > 1D are not allowed, since converting them to tuples still leaves each of its elements as unhashable lists/arrays.

I'm not sure what the best solution would be, the code in `ActiveLoop._parameter_arrays` seems quite complicated. Perhaps a different method should be used to create the setpoint arrays than simply a tuple `sp_def` per setpoint.

### Steps to reproduce
1. Create parameter that accepts custom setpoints (ArrayParameter, MultiParameter)
2. Pass along setpoints
3. Perform measurement/loop

Minimal example:
```
import numpy as np
import qcodes as qc

class TestParameter(qc.ArrayParameter):
    def __init__(self, shape=(5,), **kwargs):
        super().__init__(name='test',
                         shape=shape,
                         setpoint_names=('setpoints',),
                         setpoints=(list(np.linspace(0, 1, shape[0])),),
                         **kwargs)

    def get(self):
        return np.random.randint(0, 10, size=self.shape)

test_parameter = TestParameter()

qc.Measure(test_parameter).run('test')
```

### Expected behaviour
Measurement runs with custom setpoints

### Actual behaviour
Error is raised:
```
  File "c:\users\labuser\documents\github\silq\qcodes\qcodes\measure.py", line 74, in run
    data_set = self._dummyLoop.get_data_set(**kwargs)
  File "c:\users\labuser\documents\github\silq\qcodes\qcodes\loops.py", line 654, in get_data_set
    data_set = new_data(arrays=self.containers(), *args, **kwargs)
  File "c:\users\labuser\documents\github\silq\qcodes\qcodes\loops.py", line 463, in containers
    action_arrays = self._parameter_arrays(action)
  File "c:\users\labuser\documents\github\silq\qcodes\qcodes\loops.py", line 545, in _parameter_arrays
    if sp_def not in all_setpoints:
TypeError: unhashable type: 'list'
```

@giulioungaretti @jenshnielsen @alexcjohnson

nulinspiratie, 2017-05-29:
Temporary solution that converts all setpoints to tuples:

```
def convert_setpoints(*args):
    """
    Convert setpoints to tuples, supporting multidimensional setpoints.
    Temporary solution to make setpoints work (see issue #627).
    Args:
        *args: 1D setpoint arrays. each successive setpoints array gains an 
            extra dimension

    Returns:

    """
    if not args:
        return tuple()
    else:
        first_arg = tuple(args[0])
        remaining_args = convert_setpoints(*args[1:])
        if remaining_args:
            remaining_args = tuple((arg,) * len(first_arg)
                                  for arg in remaining_args)
        return (first_arg, ) + remaining_args
```

jenshnielsen, 2020-09-29:
This relates to the old legacy dataset. No further work on this is planed so closing this issue

closed 2020-09-29 by jenshnielsen