microsoft/Qcodes #648: Spherical coordinates

damazter, 2017-06-19:
see also: #543 
@jenshnielsen 
@MerlinSmiles 
@WilliamHPNielsen 

As discussed: My proposal for solving all spherical coordinate related issues is to write a single meta_instrument_driver which gets x, y and z parameters as input and handles the coordinate transformations internally but keeping the original phi, theta and r in its memory and setting the magnet x, y and z whenever phi, theta or r are changed.
support for tupled parameters would be a pre.
This would solve the problem for both the Oxford fridges and for the AMI magnets I think.

jdwatson, 2017-07-20:
@jenshnielsen
@MerlinSmiles
@WilliamHPNielsen
--
To the dev team - is there any update/ETA on fixing these issues with the AMI driver?

jenshnielsen, 2017-07-23:
Its certainly on my todo list but as I am the only member of the core dev team at the moment it will take a bit of time before I can get to it