microsoft/Qcodes #663: image is saved in an aspect ratio which was left as a last one

majoqubitT10, 2017-07-04:
If you encounter a bug use the following template.
If you have a feature request feel free to freestyle.
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### System
**operating system**

**qcodes branch**

**qcodes commit**

giulioungaretti, 2017-07-05:
@majoqubitT10  thanks for the report, please fill in the informations (as much as possible) according to the template. It helps us help you! 🦄

majoqubitT10, 2017-07-05:
Notes by @jenshnielsen 

Two issues 

* If plot window is rescaled it will save in the ratio, It autoscales the data but not the window at save time
* Histogram axis is not correctly rescaled
* X axis wrong in 2d conductance measurements

jenshnielsen, 2017-08-02:
This was fixed in https://github.com/qdev-dk/Qcodes/pull/54

closed 2017-08-02 by jenshnielsen