microsoft/Qcodes #668: Change QCoDeS plot axis labels from V to mV

jhenshaw, 2017-07-13:
[image]
If you encounter a bug use the following template.
If you have a feature request feel free to freestyle.
### Steps to reproduce
1. Plot something that generates an axis with 4-decimial place accuracy
2. Note that it is labeled as "0.000X" Volts (for example)

### Expected behaviour
We should simplify the axis labels to be in Mv, thus jettisoning the extra digits and de-cluttering the plot.

### Actual behaviour
Axis is labeled as "0.000X" Volts (for example)instead.
Samples are attached.

### System
Any

Master branch
[image]

**qcodes commit**

jenshnielsen, 2017-08-09:
fixed for qdev specific version in https://github.com/qdev-dk/Qcodes/pull/60

jenshnielsen, 2017-08-16:
Still need to fix the colorbar which may also have volts on the scale

closed 2017-08-17 by jenshnielsen