microsoft/Qcodes #669: Performing multiple tasks/measurements with Combined parameter breaks setpoints

jenshnielsen, 2017-07-14:
### Steps to reproduce
```python
import numpy as np

import qcodes as qc
from qcodes.tests.instrument_mocks import DummyInstrument
dac = DummyInstrument(name="dac", gates=['ch1', 'ch2'])  # The DAC voltage source

cp = qc.combine(dac.ch1, dac.ch2, name='combined_dac')

def printsomething():
    print("something")

def printelse():
    print("else")

printtask = qc.Task(printsomething)
printelse = qc.Task(printelse)

chan1_set = np.linspace(0, 10, 11).reshape(11,1)
chan2_set = np.linspace(0, 5, 11).reshape(11,1)
combine_setpoints = np.hstack((chan1_set, chan2_set))
loop = qc.Loop(cp.sweep(combine_setpoints), 0.1).each(printtask , printelse)
print(data.dac_ch1)
print(data.dac_ch2)
```
or
```python
import numpy as np

import qcodes as qc
from qcodes.tests.instrument_mocks import DummyInstrument

dac = DummyInstrument(name="dac", gates=['ch1', 'ch2'])  # The DAC voltage source
dmm = DummyInstrument(name="dmm", gates=['voltage'])  # The DMM voltage reader
import random
dmm.voltage.get =  lambda: random.randint(0, 100)

cp = qc.combine(dac.ch1, dac.ch2, name='combined_dac')

chan1_set = np.linspace(0, 10, 11).reshape(11,1)
chan2_set = np.linspace(0, 5, 11).reshape(11,1)
combine_setpoints = np.hstack((chan1_set, chan2_set))
loop = qc.Loop(cp.sweep(combine_setpoints), 0.1).each(dmm.voltage, dmm.voltage)

print(data.dac_ch1)
print(data.dac_ch2)
```
### Expected behaviour

measurements should work

### Actual behaviour
In the first case an error is raised
```
Traceback (most recent call last):
  File "C:/Users/Jens/Majorana/combined_example.py", line 31, in <module>
    data = loop.run()
  File "C:\Users\Jens\Qcodes\qcodes\loops.py", line 735, in run
    self._run_wrapper()
  File "C:\Users\Jens\Qcodes\qcodes\loops.py", line 784, in _run_wrapper
    self._run_loop(*args, **kwargs)
  File "C:\Users\Jens\Qcodes\qcodes\loops.py", line 840, in _run_loop
    set_name = (self.data_set.action_id_map[set_index])
KeyError: (1,)
```
Furthermore the setpoints stored in dac_ch1 and dac_ch2 are wrong
(first array contains values from 2nd and 2en empty)

### System
Windows 
QCodes master

MerlinSmiles, 2017-07-14:
@jenshnielsen we ran into the same...
in loops.py this:

``` Python
            if hasattr(self.sweep_values, "parameters"):
                set_name = self.data_set.action_id_map[action_indices]
                if hasattr(self.sweep_values, 'aggregate'):
                    value = self.sweep_values.aggregate(*set_val)
                self.data_set.store(new_indices, {set_name: value})
                self.publisher.store(new_indices, {set_name: value}, uuid=self.data_set.uuid)
                # for j, val in enumerate(set_val):
                #     set_index = action_indices + (j+1, )
                #     set_name = (self.data_set.action_id_map[set_index])
                #     data_to_store[set_name] = val
            else:
                set_name = self.data_set.action_id_map[action_indices]
                data_to_store[set_name] = value
```

made me forget the issue 😮 
there is something wrong with that indexing, that j is just randomly counting and has nothing to do with the actual action index that parameter got.

jenshnielsen, 2017-07-14:
Yes that also made me want to forget it ;)

jenshnielsen, 2017-07-14:
@MerlinSmiles noticed that this leads to unsaved data. This happens because this bug results in lines that contains nan's and the match_save_range check  https://github.com/QCoDeS/Qcodes/blob/master/qcodes/data/gnuplot_format.py#L269 only writes complete lines

MerlinSmiles, 2017-07-14:
lol great that you found the reason, I didnt figure that part out

PR #670 (2017-07-18): Correctly count number of actions when updating combined measurements
Fixes #669 

The issue is that the base parameters within the combined parameter were being looked up from an index that always starts at 1. This is only true if there is only one parameter/task being measured/executed within the loop. This corrects the counting

- [ ] Print/log warning in case data is dropped
- [ ] Tests!
closed 2017-07-18 by jenshnielsen