microsoft/Qcodes #673: ImportError: Cannot import name "InstrumentChannel" and "ChannelList"

ArvindK12345, 2017-07-17:
If you encounter a bug use the following template.
If you have a feature request feel free to freestyle.
### Steps to reproduce
1. from qcodes import InstrumentList, ChannelList
2.
3.

### Expected behaviour
Tell us what should happen

The modules should import and allow me to create a driver for a new Oscilloscope

### Actual behaviour
Tell us what happens instead

The modules don't import and I receive the error message as in the title.

### System
**operating system**
Windows 7 Pro, Service Pack 1
Intel i7

**qcodes branch**

**qcodes commit**

I am trying to write a driver for a new oscilloscope (LeCroy WaveRunner 104xi) based on the existing driver for the Tektronix TPS2012 oscilloscope (qcodes.instrument_drivers.tektronix.TPS2012 module) for reflectometry measurements. While trying to import the LeCroy driver, I receive the error message given in the thread title. Is this issue something you have come across? How do I go about fixing this?

The following is the code for the LeCroy driver (still unfinished but I want to see if it works so far):

#!/usr/bin/env python
# -*- coding: utf-8 -*-

import logging
import binascii
import qcodes

import numpy as np
from pyvisa.errors import VisaIOError
from functools import partial

from qcodes import *
from qcodes import VisaInstrument, validators as vals
from qcodes import InstrumentChannel, ChannelList
from qcodes import ArrayParameter

log = logging.getLogger(__name__)

class TraceNotReady(Exception):
    pass

class ScopeArray(ArrayParameter):
    def __init__(self, name, instrument, channel):
        super().__init__(name=name,
                         shape=(2500,),
                         label='Voltage',
                         unit='V ',
                         setpoint_names=('Time', ),
                         setpoint_labels=('Time', ),
                         setpoint_units=('s',),
                         docstring='holds an array from scope')
        self.channel = channel
        self._instrument = instrument

    def calc_set_points(self):
        message = self._instrument.ask('WAVE_ARRAY_COUNT?')
        preamble = self._preambleparser(message)
        xstart = preamble['x_zero']
        xinc = preamble['x_incr']
        no_of_points = preamble['no_of_points']
[... 292 lines cut ...]
    def clear_message_queue(self, verbose=False):
        """
        Function to clear up (flush) the VISA message queue of the AWG
        instrument. Reads all messages in the the queue.

        Args:
            verbose (Bool): If True, the read messages are printed.
                Default: False.
        """
        original_timeout = self.visa_handle.timeout
        self.visa_handle.timeout = 1000  # 1 second as VISA counts in ms
        gotexception = False
        while not gotexception:
            try:
                message = self.visa_handle.read()
                if verbose:
                    print(message)
            except VisaIOError:
                gotexception = True
        self.visa_handle.timeout = original_timeout

Please let me know if there is a fix for this.

Many Thanks,
Arvind

jenshnielsen, 2017-07-17:
Hi Arvind,
I assume that you mean:
```
from qcodes import InstrumentChannel, ChannelList
```
and not ```InstrumentList``` I.e your title of the bug is correct but there is a typo in your list below? 

The InstrumentChannels and ChannelList was added in the latest version of QCoDeS along with a new version of the TPS Driver that uses the channels. How did you install qcodes? and do you have the latest version?

ArvindK12345, 2017-07-17:
Hi Jens,

Thanks for the quick reply. I installed qCoDeS via "pip install qcodes" but that was several months ago, has a new version come out since? 
You are correct about the typo in the list below, I'll change that now.

jenshnielsen, 2017-07-17:
Yes we did a release in both June and July. The Current version of the TPS driver (Which I guess you found here on Github?) only works with the latest release as it uses new features not in the old releases

ArvindK12345, 2017-07-17:
I've updated my one to the latest version of qCoDeS via "pip install --upgrade qcodes" but it's still giving me the same error. Do you know why this might be the case?
And you are right, I found the current version of the TPS driver on Github.

jenshnielsen, 2017-07-17:
Can you double check that you have the latest version by doing 

```
import qcodes
print(qcodes.version.__version__)
```
from a python prompt. That should print 0.1.5

ArvindK12345, 2017-07-17:
It's still on 0.1.3 however it says the latest version has been installed on cmd.exe
Is there a step I am missing between pip install/upgrade and being able to use the latest version?

jenshnielsen, 2017-07-17:
There should not be. Is it possible that you have multiple python installations and you have upgraded the version installed in one and is using the other. Does the output of `pip list` show the right version?

ArvindK12345, 2017-07-17:
Yes, that was exactly the error. I upgraded qCoDeS on the correct Python installation and now can import/use the new TPS driver.

Thanks a lot for the help, I appreciate it.

Cheers,
Arvind

jenshnielsen, 2017-07-17:
Great I will close this then

closed 2017-07-17 by jenshnielsen