microsoft/Qcodes #674: Oscilloscope Templates

ArvindK12345, 2017-07-17:
Hey,

Do you have a template which you follow for creating a driver for an oscilloscope?

Cheers,
Arvind

jenshnielsen, 2017-07-17:
Not really, I don't think we have a lot of Oscilloscope driver. The TPS driver is probably the best  starting point at the moment but if you have any specific questions feel free to ask me

ArvindK12345, 2017-07-17:
I'm afraid I don't actually understand what you are trying to do with the _binaryparser, _preambleparser and  _curveparameterparser functions in the ScopeArray Class and the entirety of the TPS2012Channel class. Plus it seems as though there's isn't a similar preamble for the oscilloscope I'm trying to use (LeCroy Waverunner 104Xi) so I'm unsure as to how to replicate this.

jenshnielsen, 2017-07-18:
The Channel groups all parameters that relate to a specific channel. From QCoDeS 0.1.5 we have added that concept so that you can group the parameters in a logical way and do instrument.channels.ch1.parameter rather than instrument.ch1_parameter and so on. This makes it much easier to write code that can do the same operations for multiple channels such as an oscilloscope that typically has at least 2 identical channels. 

The ScopeArray is an array parameter. This needs special treatment because the scope returns an array of data representing the time series. The time series has a variable length, offset and time resolution depending on the settings of the scope. In this particular scope you need to get the preamble to find the data to calculate the x-axis array

ArvindK12345, 2017-07-18:
So does that mean if we have a scope where we don't need to use the preamble and if we wanted to leave the syntax as ch1_parameter, etc. that only the bottom class needs to be replicated? i.e. recreating this should suffice:

https://github.com/WilliamHPNielsen/Qcodes/blob/280aea9604c1ebf1527e700c58e976922fd82b11/qcodes/instrument_drivers/tektronix/TPS2012.py

Cheers.

jenshnielsen, 2017-07-18:
Yes you don't have to use channels but I would strongly discourage not to use them. It leads to unstructured and much harder to use. We will be gradually converting all drivers shipped with QCoDeS to use channels where it makes sense. As soon as you write code with the old pattern and want to switch channel you will end up writing ugly code such as `getattr(inst, 'ch{}_parameter'.format(channel_number).get()` and so on this is so much cleaner with channels. 

Using the Channels is not that much of an additional complication it's mainly a matter of moving all the channel specific parameters to the channel class and then add the channels to the instrument with code like the following

```python
        # channel-specific parameters
        channels = ChannelList(self, "ScopeChannels", TPS2012Channel, snapshotable=False)
        for ch_num in range(1, 3):
            ch_name = "ch{}".format(ch_num)
            channel = TPS2012Channel(self, ch_name, ch_num)
            channels.append(channel)
        channels.lock()
        self.add_submodule("channels", channels)
```

You do currently need to subclass ArrayParameter as soon as you want a parameter that returns an array, this is because you manually need to specify how to load the data and the x-axis.
 
But that is not to say that you cannot start just with a Instrument class and implement all the StandardParameter, such as setting horizontal axis and so on and then implement the Channels and Arrayparameters later on

ArvindK12345, 2017-07-19:
Okay so I've managed to get it working using the instrument class and will add the channel and arrayparameters later. With regards to the array output, is there a way to save this data while looping through numerous runs?

Cheers!

closed 2017-07-20 by ArvindK12345