microsoft/Qcodes #678: "ValueError: setting an array element with a sequence." on oscilloscope driver

ArvindK12345, 2017-07-20:
Is there a specific format when trying to saving data using the measure.Measure.get() command using an oscilloscope? I have tried using a numpy array, a list and an object but they all give me the same error:

"ValueError: setting an array element with a sequence."

I know this was an issue you came across in #435 . How did you resolve this error in the TPS driver?

Cheers!
Arvind

jenshnielsen, 2017-07-21:
Hi Arvind 

You will need to use a subclass of an arrayparameter for this to work as discussed in #674

ArvindK12345, 2017-07-21:
Did you ever run into the following error when making the TPS2012 driver? If so, how did you go about fixing it?

import qcodes as qc
lcr.ch1_curvedata.prepare_curvedata()
#data = qc.Measure(lcr.ch1_curvedata).run()

AttributeError                            Traceback (most recent call last)
<ipython-input-3-82cd0f153431> in <module>()
      1 import qcodes as qc
----> 2 lcr.ch1_curvedata.prepare_curvedata()
      3 #data = qc.Measure(lcr.ch1_curvedata).run()

~\Anaconda3\envs\QCODES\lib\site-packages\qcodes\utils\helpers.py in __getattr__(self, key)
    356         raise AttributeError(
    357             "'{}' object and its delegates have no attribute '{}'".format(
--> 358                 self.__class__.__name__, key))
    359 
    360     def __dir__(self):

AttributeError: 'LCR' object and its delegates have no attribute 'ch1_curvedata'

I have also attached my in-progress driver for the LeCroy oscilliscope, could you have a look and see where I've made an error with the curvedata function?

Thanks!!

``
import logging
import binascii

import numpy as np
from pyvisa.errors import VisaIOError
from functools import partial

from qcodes import VisaInstrument, validators as vals
from qcodes import InstrumentChannel, ChannelList
from qcodes import ArrayParameter

log = logging.getLogger(__name__)

class TraceNotReady(Exception):
    pass

class ScopeArray(ArrayParameter):
    def __init__(self, name, instrument, channel):
        super().__init__(name=name,
                         shape=(2500,),
                         label='Voltage',
                         unit='V ',
                         setpoint_names=('Time', ),
                         setpoint_labels=('Time', ),
                         setpoint_units=('s',),
                         docstring='holds an array from scope')
        self.channel = channel
        self._instrument = instrument

    def calc_set_points(self):
        
        xstart = self.instrument.ask("VBS? 'Return=app.Acquisition.Channels(\"C{}\")".format(ch) + ".Out.Result.HorizontalOffset'")  # Retrieve X zero
        xincr = self.instrument.ask("VBS? 'Return=app.Acquisition.Channels(\"C{}\")".format(ch) + ".Out.Result.HorizontalPerStep'")  # Retrieve X incr
        ptcnt = self.instrument.ask("VBS? 'Return=app.Acquisition.Channels(\"C{}\")".format(ch) + ".Out.Result.Samples'")  # Retrieve number of data points      
        

        xdata = np.linspace(xstart, no_of_points * xinc + xstart, no_of_points)
        return xdata, no_of_points

    def prepare_curvedata(self):
        """
        Prepare the scope for returning curve data
        """
        # To calculate set points, we must have the full preamble
[... 185 lines cut ...]
    # METHODS FOR THE USER                           #
    ##################################################

    def clear_message_queue(self, verbose=False):
        """
        Function to clear up (flush) the VISA message queue of the AWG
        instrument. Reads all messages in the the queue.

        Args:
            verbose (Bool): If True, the read messages are printed.
                Default: False.
        """
        original_timeout = self.visa_handle.timeout
        self.visa_handle.timeout = 1000  # 1 second as VISA counts in ms
        gotexception = False
        while not gotexception:
            try:
                message = self.visa_handle.read()
                if verbose:
                    print(message)
            except VisaIOError:
                gotexception = True
        self.visa_handle.timeout = original_timeout

>

jenshnielsen, 2017-07-21:
Your LCR class has no attribute called ch1_curvedata anymore when you use the channels.

It will be 

```
lcr.channels.ch1.curvedata
``
or
```
lcr.channels[0].curvedata
```

jenshnielsen, 2017-07-21:
Unfortunately the example notebook uses the old pre channels interface and should be updated

ArvindK12345, 2017-07-22:
Oh sure, that seems to work now. Cheers for the help Jens.
On a slightly different note, how do I go about submitting drivers for a few new instruments?

jenshnielsen, 2017-07-23:
If you create a github pull request with the driver that you have created we can review it and give feedback

Dominik-Vogel, 2017-10-03:
Hey @ArvindK12345!
I was wondering about the status of your oscilloscope driver. It would be great to share it with other QCoDeS users. Could you make a pull request?
If you need help with anything just let me know.

ArvindK12345, 2017-10-12:
Hey Dominik,

Apologies for the delay but I have now created pull requests for the oscilloscope driver (https://github.com/ArvindK12345/Qcodes/pull/2) and a microwave source driver (https://github.com/ArvindK12345/Qcodes/pull/1). Could you merge them with the main QCoDeS branch when you get some time?

Cheers!

jenshnielsen, 2017-10-13:
Hi Arvind, 

It looks like you have made those pull requests against your own fork of qcodes. Can you open them against the qcodes/qcodes main repository. I am happy to explain more if you need it

ArvindK12345, 2017-10-13:
You were right, I should have sorted that out now. Could you check if they're merged?

WilliamHPNielsen, 2017-10-16:
Hi @ArvindK12345, it looks like you could use a helping hand in making this pull request. I have sent you an invitation to join the QCoDeS slack channel; there you can get more comprehensive step-by-step support.

ArvindK12345, 2017-10-17:
> Hi @ArvindK12345 [1], it looks like you could use a helping hand in
> thread [3].
> Links:

Hey Will,

You are right I've been having some trouble with the pull request. I've 
sent you the files I've been trying to upload since I'll be busy for the 
next few days. Could you please upload these and I'll try and work 
through any improvements afterwards.

Cheers,
Arvind