microsoft/Qcodes #694: Keithley 2614b visa issues

majoqubitT10, 2017-08-04:
Frequent connect and disconnect of Keithley Source meter results in Visa errors

WilliamHPNielsen, 2017-10-03:
Since the description of the problem is so vague, it is very hard to come up with any means by which to resolve the problem. Unless we get a more detailed bug report, we might as well close this issue.

Dominik-Vogel, 2017-10-03:
@WilliamHPNielsen I just saw this issue, when I was doing a clean up and though of you as an expert on the Keithley... But I agree from the description of the error there is nothing one can do...

jenshnielsen, 2017-10-03:
The bug report is written by me

jenshnielsen, 2017-10-03:
It's quite simple the Keithley is very likely to timeout if you kill the kernel but dont disconnect.
There is not much we can do apart from the atexit trick

astafan8, 2018-06-14:
@jenshnielsen I could not understand the problem completely, but it seems you know the fix and it seems quick. Could you at least write it down here? Or, if you have time, make a PR?

WilliamHPNielsen, 2018-06-15:
This is not as such a bug in QCoDeS.

The problem is that some instruments, the Keithley for instance, do not handle dead/multiple connections very well. That means that the user *must* make sure to explicitly close any connection made to that instrument, or new connection will not be accepted. The fix to ensure that this always happens is to use the `atexit` module to close all QCoDeS `Instrument` instances. This is done in `__init__.py`, here and the next line: https://github.com/QCoDeS/Qcodes/blob/ac88bd6f7071fc0861fa3cbf2866f3a6e0db1bb8/qcodes/__init__.py#L93

So, yeah, I guess this issue is resolved?

jenshnielsen, 2018-06-15:
It's not resolved but worked around but short of patching the firmware there is not much more we can do

closed 2018-06-15 by jenshnielsen