microsoft/Qcodes #705: Driver: Yokogawa GS210

jenshnielsen, 2017-08-11:
I don't find that model online:

Could be the GS200 or the GS610 perhaps
http://tmi.yokogawa.com/vn/products/generators-sources/source-measure-units/

We have a very minimal driver for the GS200 in https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument_drivers/yokogawa/GS200.py

jenshnielsen, 2017-08-14:
More information from John Watson
```
The GS210 is just the GS200 with the option of having the output terminals on the front of the instrument (Yokogawa’s documentation is kind of confusing on this), so the GS200 driver is the correct one for what we are buying.  I took a quick look at the driver, and I think it will need to be extended a bit to allow the types of measurements we typically would use it for (it looks like it currently just initializes communication with the instrument).```

spauka, 2017-09-14:
Coincidentally I've been working on this driver at the same time.

See:
https://github.com/spauka/Qcodes/blob/device-dev/qcodes/instrument_drivers/yokogawa/GS200.py

It will be ready for a pull request soon...

jenshnielsen, 2017-09-14:
@spauka Great and thanks for the heads up

sohailc, 2017-11-02:
This issue belong on the StationQ repo: https://github.com/QCoDeS/StationQ/issues/16

closed 2017-11-02 by sohailc