microsoft/Qcodes #709: Spyder: Interrupt multiple scripts

jenshnielsen, 2017-08-14:
If multiple scripts are being run it's not by pressing the green play button multiple times it's not possible to interrupt the second one. This may be a bug in spyder that is fixed in the latest release

astafan8, 2018-06-14:
@jenshnielsen why is a Spyder issue mentioned in QCoDeS?

astafan8, 2018-06-15:
@jenshnielsen then I would still open this issue in Spyder's github instead of here.

in the meantime, I coulkd reproduce the problem:

1. write dummy script
```
from time import sleep

for i in range(5):
    print(i)
    sleep(1)
    
print('done')
```
2. Set up the run configuration (Run -> Configurations per file): select "Execute in current console"
3. Run the file by pressing the green "Run file" button, let it print one or two numbers, and then press the run button again.
4. Immediately, press the red "Stop" button in the console window.
5. Both scripts should be interrupted (as expected)
6. Now let's change the run configuration to "execute in a dedicated console"
7. Run the file by pressing the green "Run file" button, let it print one or two numbers, and then press the run button again.
8. Immediately, press the red "Stop" button in the console window. The first script will be interrupted, but the second one will still start its execution and it will start printing its numbers like this:
```
runfile('C:/Users/a-miasta/.spyder-py3/temp.py', wdir='C:/Users/a-miasta/.spyder-py3')
0
1

2
Traceback (most recent call last):

  File "<ipython-input-9-18c95d7ea4c8>", line 1, in <module>
    runfile('C:/Users/a-miasta/.spyder-py3/temp.py', wdir='C:/Users/a-miasta/.spyder-py3')

  File "C:\Users\a-miasta\AppData\Local\Continuum\anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 705, in runfile
    execfile(filename, namespace)

  File "C:\Users\a-miasta\AppData\Local\Continuum\anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/a-miasta/.spyder-py3/temp.py", line 12, in <module>
    sleep(1)

KeyboardInterrupt

runfile('C:/Users/a-miasta/.spyder-py3/temp.py', wdir='C:/Users/a-miasta/.spyder-py3')
ERROR: execution aborted
0
1
2
3
4
done
```

9. What is even worse, is that the red "Stop" button becomes inactive after you press it first time to stop the first run.

jenshnielsen, 2018-06-15:
>  @jenshnielsen then I would still open this issue in Spyder's github instead of here.
Obviously! but to do that you would need to probably reproduce it! This was just a quick note to remind us to get back to it and investigate more. Which we failed to do

jenshnielsen, 2018-06-15:
Now that you have reproduces it, it would be great to open a real issue

astafan8, 2018-06-15:
😸 for a bit more than a year, no one has successfully been reminded about this issue 😸 

Reproduced it and created an issue in Spyder: https://github.com/spyder-ide/spyder/issues/7293 😈

closed 2018-06-15 by astafan8