microsoft/Qcodes #728: Threading exception from live plotting and changes prompt font to black

lampertl, 2017-09-07:
### Steps to reproduce
1. Run live plot (QtPlot). In this case, for debugging purposes, I ran the 1D loop example (https://github.com/QCoDeS/Qcodes/blob/master/docs/examples/Tutorial.ipynb) with live plotting and it throws a threading exception.
2. Live plot displays and runs correctly during data acquisition
3. After live plot ends upon sweep completion, Python throws an exception and changes to font to black in the anaconda prompt window.

### Expected behaviour
The live plot should complete without throwing an exception. In some cases, the threading exception is thrown before the rest of the sequence can be completed.

### Actual behaviour
The exception thrown is:
Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Users\llamper\AppData\Local\Continuum\Anaconda3\envs\Python_3_5_ENV\lib\threading.py", line 914, in _bootstrap_inner
    self.run()
  File "C:\Users\llamper\AppData\Local\Continuum\Anaconda3\envs\Python_3_5_ENV\lib\site-packages\pyqtgraph\multiprocess\processes.py", line 469, in run
    cprint.cerr(self.color, line, -1)
  File "C:\Users\llamper\AppData\Local\Continuum\Anaconda3\envs\Python_3_5_ENV\lib\site-packages\pyqtgraph\util\cprint.py", line 100, in cerr
    cprint('stderr', *args)
  File "C:\Users\llamper\AppData\Local\Continuum\Anaconda3\envs\Python_3_5_ENV\lib\site-packages\pyqtgraph\util\cprint.py", line 79, in cprint
    kwds = WIN[arg]
KeyError: b'C:\\Users\\llamper\\AppData\\Local\\Continuum\\Anaconda3\\envs\\Python_3_5_ENV\\lib\\site-packages\\pyqtgraph\\graphicsItems\\PlotCurveItem.py:131: RuntimeWarning: All-NaN axis encountered\r\n'

Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\llamper\AppData\Local\Continuum\Anaconda3\envs\Python_3_5_ENV\lib\threading.py", line 914, in _bootstrap_inner
    self.run()
  File "C:\Users\llamper\AppData\Local\Continuum\Anaconda3\envs\Python_3_5_ENV\lib\site-packages\pyqtgraph\multiprocess\processes.py", line 464, in run
    cprint.cout(self.color, line, -1)
  File "C:\Users\llamper\AppData\Local\Continuum\Anaconda3\envs\Python_3_5_ENV\lib\site-packages\pyqtgraph\util\cprint.py", line 96, in cout
    cprint('stdout', *args)
  File "C:\Users\llamper\AppData\Local\Continuum\Anaconda3\envs\Python_3_5_ENV\lib\site-packages\pyqtgraph\util\cprint.py", line 79, in cprint
    kwds = WIN[arg]
KeyError: b''

### System
Win 10

Master

jenshnielsen, 2017-09-08:
Thanks for the report .

Can you elaborate a bit more on how you run this. Are you running the Tutorial from the jupyter notebook or copying it to a prompt. 
Is it the syntax highlighting of for code that changes to black or the output of running the notebook?
Does this happen every time or randomly. I cannot reproduce it on my setup.
The output of `conda list` and `pip list` to see your installed packages would be useful too.

lampertl, 2017-09-08:
Thanks for the quick reply!

Some quick background: I was creating some scripts to run some qcodes for a measurement setup and this thread exception was thrown after live plots were completed (the code has no issues when live plotting is disabled). I then ran the tutorial live plot to determine if the live plotting feature was indeed the issue (testing the tutorial was recommended to me). To run the live plot tutorial, I copied the code into my text editor, saved to my local machine, and executed the code via anaconda environment prompt.

The output of the anaconda prompt window is normally black background and white text. When the threading exception is thrown immediately after a live plot is completed, the text changes to black as well so it cannot be seen on the anaconda prompt. I copy/paste the text into a text editor so I can read the error message (same as posted in my original report). 

This happens every time I run live plots. The live plots work, then after completing a scan and closing, the exception is thrown. 

conda list:

bleach                    2.0.0                     <pip>
bzip2                     1.0.6                    vc14_3  [vc14]  defaults
colorama                  0.3.9                     <pip>
cycler                    0.10.0                   py35_0    defaults
cycler                    0.10.0                    <pip>
decorator                 4.1.2                     <pip>
decorator                 4.1.2                    py35_0    defaults
entrypoints               0.2.3                     <pip>
freetype                  2.5.5                    vc14_2  [vc14]  defaults
gevent                    1.2.2                     <pip>
graphviz                  2.38.0                        4    defaults
graphviz                  0.8                       <pip>
greenlet                  0.4.12                    <pip>
h5py                      2.7.0               np113py35_0    defaults
hdf5                      1.8.15.1                 vc14_4  [vc14]  defaults
hickle                    2.1.0                     <pip>
html5lib                  0.999999999               <pip>
icu                       57.1                     vc14_0  [vc14]  defaults
ipykernel                 4.6.1                     <pip>
ipython                   6.1.0                     <pip>
ipython-genutils          0.2.0                     <pip>
ipywidgets                7.0.0                     <pip>
jedi                      0.10.2                    <pip>
Jinja2                    2.9.6                     <pip>
jpeg                      9b                       vc14_0  [vc14]  defaults
jsonschema                2.6.0                     <pip>
jupyter                   1.0.0                     <pip>
jupyter-client            5.1.0                     <pip>
jupyter-console           5.2.0                     <pip>
jupyter-core              4.3.0                     <pip>
libpng                    1.6.30                   vc14_1  [vc14]  defaults
libtiff                   4.0.6                    vc14_3  [vc14]  defaults
MarkupSafe                1.0                       <pip>
matlabengineforpython     R2017a                    <pip>
matplotlib                2.0.2               np113py35_0    defaults
matplotlib                2.0.2                     <pip>
mistune                   0.7.4                     <pip>
mkl                       2017.0.3                      0    defaults
nbconvert                 5.3.0                     <pip>
nbformat                  4.4.0                     <pip>
networkx                  1.11                     py35_0    defaults
notebook                  5.0.0                     <pip>
numpy                     1.13.1                   py35_0    defaults
olefile                   0.44                     py35_0    defaults
opencv                    1.0.1                     <pip>
opencv-python             3.3.0                     <pip>
openssl                   1.0.2l                   vc14_0  [vc14]  defaults
pandocfilters             1.4.2                     <pip>
pickleshare               0.7.4                     <pip>
pillow                    4.2.1                    py35_0    defaults
pip                       9.0.1                    py35_1    defaults
prompt-toolkit            1.0.15                    <pip>
Pygments                  2.2.0                     <pip>
pyparsing                 2.2.0                     <pip>
pyparsing                 2.2.0                    py35_0    defaults
pyqt                      5.6.0                    py35_2    defaults
PyQt5                     5.9                       <pip>
pyqtgraph                 0.10.0                    <pip>
python                    3.5.4                         0    defaults
python-dateutil           2.6.1                     <pip>
[... 76 lines cut ...]
python-dateutil (2.6.1)
pytz (2017.2)
PyVISA (1.8)
PyWavelets (0.5.2)
pyzmq (16.0.2)
qcodes (0.1.6)
qtconsole (4.3.1)
QtPy (1.3.1)
qtt (0.1.3)
qttstations (0.1.0)
redis (2.10.6)
scikit-image (0.13.0)
scipy (0.19.1)
setuptools (27.2.0)
simplegeneric (0.8.1)
sip (4.19.3)
six (1.10.0)
testpath (0.3.1)
tornado (4.5.2)
traitlets (4.3.2)
wcwidth (0.1.7)
webencodings (0.5.1)
wheel (0.29.0)
widgetsnbextension (3.0.2)
win-unicode-console (0.5)

jenshnielsen, 2017-09-08:
one thing i noticed is that you have both pyqt5.6 and 5.9 installed likely from conda and pip respectively. We have previously seen that leading to odd behaviour similar to yours. 

You can try creating a fresh env from the ` environment.yml` file in the root of the qcodes repository with `conda env create -f environment.yml`

lampertl, 2017-09-08:
Is there a way I can run qcodes without having both installed? I find that when only one is installed, I have a matplotlib issue. At the moment, I need both pyqt and pyqt5 to run qcodes.

Is there an advantage to creating a fresh environment within the root of the qcodes repository? I was intending to run qcodes at the measurement station using the anaconda prompt.

jenshnielsen, 2017-09-08:
You cant have both installed. It's the really the same package so that is going to leave you with a broken mixture of both. QCodes should work fine with either. I suggest that you set up an environment with only conda packages as I suggest above and we then focus on resolving any issue you have in that

jenshnielsen, 2017-09-08:
Looking at your lists you might also have matplotlib installed twice which may be the root cause for your isses

lampertl, 2017-09-08:
OK, thank you very much for the feedback. I will create a fresh environment with only conda packages and ensure none of them are doubled. I'm traveling abroad for 2 weeks so I may not follow up on this very much until I return.

lampertl, 2017-09-26:
I returned and installed a fresh environment being careful not to duplicate any packages and sticking with conda packages where possible. After running the live plotting, I receive the same threading exception as before. Here is conda list from the fresh environment:

bleach                    1.5.0                    py35_0
bzip2                     1.0.6                    vc14_3  [vc14]  anaconda
certifi                   2016.2.28                py35_0
colorama                  0.3.9                    py35_0
cycler                    0.10.0                   py35_0
decorator                 4.1.2                    py35_0
entrypoints               0.2.3                    py35_0
freetype                  2.5.5                    vc14_2  [vc14]  anaconda
h5py                      2.7.1                     <pip>
hickle                    2.1.0                     <pip>
html5lib                  0.9999999                py35_0
icu                       57.1                     vc14_0  [vc14]
ipykernel                 4.6.1                    py35_0
ipython                   6.1.0                    py35_0
ipython_genutils          0.2.0                    py35_0
ipywidgets                6.0.0                    py35_0
jedi                      0.10.2                   py35_2
jinja2                    2.9.6                    py35_0
jpeg                      9b                       vc14_0  [vc14]
jsonschema                2.6.0                    py35_0
jupyter                   1.0.0                    py35_3
jupyter_client            5.1.0                    py35_0
jupyter_console           5.2.0                    py35_0
jupyter_core              4.3.0                    py35_0
libpng                    1.6.30                   vc14_1  [vc14]
libsodium                 1.0.3                         0
libtiff                   4.0.6                    vc14_3  [vc14]  anaconda
markupsafe                1.0                      py35_0
matplotlib                2.0.2               np113py35_0
mistune                   0.7.4                    py35_0
mkl                       2017.0.3                      0
nbconvert                 5.2.1                    py35_0
nbformat                  4.4.0                    py35_0
networkx                  1.11                     py35_0    anaconda
notebook                  5.0.0                    py35_0
numpy                     1.13.1                    <pip>
numpy                     1.13.1                   py35_0
olefile                   0.44                     py35_0    anaconda
opencv-python             3.3.0                     <pip>
openssl                   1.0.2l                   vc14_0  [vc14]
pandocfilters             1.4.2                    py35_0
path.py                   10.3.1                   py35_0
pickleshare               0.7.4                    py35_0
pillow                    4.2.1                    py35_0    anaconda
pip                       9.0.1                    py35_1
prompt_toolkit            1.0.15                   py35_0
pygments                  2.2.0                    py35_0
pyparsing                 2.2.0                    py35_0
pyqt                      5.6.0                    py35_2
pyqtgraph                 0.10.0                   py35_0
python                    3.5.4                         0
python-dateutil           2.6.1                    py35_0
pytz                      2017.2                   py35_0
PyVISA                    1.8                       <pip>
pywavelets                0.5.2               np113py35_0    anaconda
pyzmq                     16.0.2                   py35_0
qcodes                    0.1.7                     <pip>
qt                        5.6.2                    vc14_6  [vc14]
qtconsole                 4.3.1                    py35_0
qtpy                      1.3.1                    py35_0
qtt                       0.1.3                     <pip>
scikit-image              0.13.0              np113py35_0    anaconda
scipy                     0.19.1              np113py35_0
setuptools                36.4.0                   py35_1
simplegeneric             0.8.1                    py35_1
sip                       4.18                     py35_0
six                       1.10.0                   py35_1
six                       1.11.0                    <pip>
testpath                  0.3.1                    py35_0
tk                        8.5.18                   vc14_0  [vc14]
tornado                   4.5.2                    py35_0
traitlets                 4.3.2                    py35_0
vs2015_runtime            14.0.25420                    0
wcwidth                   0.1.7                    py35_0
wheel                     0.29.0                   py35_0
widgetsnbextension        3.0.2                    py35_0
win_unicode_console       0.5                      py35_0
wincertstore              0.2                      py35_0
zeromq                    4.1.3                    vc14_1  [vc14]
zlib                      1.2.11                   vc14_0  [vc14]

jenshnielsen, 2017-09-26:
There are still some odd looking duplicate packages could you try putting 

```
name: qcodes-minimal
channels:
 - defaults
 - conda-forge
dependencies:
 - h5py
 - matplotlib
 - pyqtgraph
 - python=3.6
 - jsonschema
 - pyvisa
 - QtPy
 - sip
 - pyqt=5.6.0
 - numpy
 - jupyter=1.0.0
 - typing
 - hypothesis
 - pytest
 - pytest-runner
```
in a file called `env.yml` and then creating an environment with 
```
conda env create -f env.yml
```
activate the env and only installing qcodes in that env with `pip install qcodes` and then run the notebook from that env. 

conda list for the env should be 

```
$ conda list
# packages in environment at C:\Users\Jens\Anaconda3\envs\qcodes:
#
bleach                    1.5.0                    py36_0
certifi                   2016.2.28                py36_0
colorama                  0.3.9                    py36_0
cycler                    0.10.0                   py36_0
decorator                 4.1.2                    py36_0
entrypoints               0.2.3                    py36_0
h5py                      2.7.0               np113py36_0
hdf5                      1.8.15.1                 vc14_4  [vc14]
html5lib                  0.9999999                py36_0
hypothesis                3.23.0                   py36_0    conda-forge
icu                       57.1                     vc14_0  [vc14]
ipykernel                 4.6.1                    py36_0
ipython                   6.1.0                    py36_0
ipython_genutils          0.2.0                    py36_0
ipywidgets                6.0.0                    py36_0
jedi                      0.10.2                   py36_2
jinja2                    2.9.6                    py36_0
jpeg                      9b                       vc14_0  [vc14]
jsonschema                2.6.0                    py36_0
jupyter                   1.0.0                    py36_3
jupyter_client            5.1.0                    py36_0
jupyter_console           5.2.0                    py36_0
jupyter_core              4.3.0                    py36_0
libpng                    1.6.30                   vc14_1  [vc14]
markupsafe                1.0                      py36_0
matplotlib                2.0.2               np113py36_0
mistune                   0.7.4                    py36_0
mkl                       2017.0.3                      0
nbconvert                 5.2.1                    py36_0
nbformat                  4.4.0                    py36_0
notebook                  5.0.0                    py36_0
numpy                     1.13.1                   py36_0
openssl                   1.0.2l                   vc14_0  [vc14]
pandocfilters             1.4.2                    py36_0
path.py                   10.3.1                   py36_0
pickleshare               0.7.4                    py36_0
pip                       9.0.1                    py36_1
prompt_toolkit            1.0.15                   py36_0
py                        1.4.34                   py36_0
pygments                  2.2.0                    py36_0
pyparsing                 2.2.0                    py36_0
pyqt                      5.6.0                    py36_2
pyqtgraph                 0.10.0                   py36_0
pytest                    3.2.1                    py36_0
pytest-runner             2.11.1                   py36_0
python                    3.6.2                         0
python-dateutil           2.6.1                    py36_0
pytz                      2017.2                   py36_0
pyvisa                    1.8                      py36_0    conda-forge
pyzmq                     16.0.2                   py36_0
qcodes                    0.1.7                     <pip>
qt                        5.6.2                    vc14_6  [vc14]
qtconsole                 4.3.1                    py36_0
qtpy                      1.3.1                    py36_0
setuptools                36.4.0                   py36_1
simplegeneric             0.8.1                    py36_1
sip                       4.18                     py36_0
six                       1.10.0                   py36_0
testpath                  0.3.1                    py36_0
tk                        8.5.18                   vc14_0  [vc14]
tornado                   4.5.2                    py36_0
traitlets                 4.3.2                    py36_0
typing                    3.6.2                    py36_0
vs2015_runtime            14.0.25420                    0
wcwidth                   0.1.7                    py36_0
wheel                     0.29.0                   py36_0
widgetsnbextension        3.0.2                    py36_0
wincertstore              0.2                      py36_0
zlib                      1.2.11                   vc14_0  [vc14]

```

I suspect that there is some incompatibility with one of the other packages that you have installed. Lets see if we can track it down

lampertl, 2017-09-26:
Thank you for the feedback! I followed exactly what you said and still receive the same threading exception even running the 1D loop example. 

Here is conda list:

`# packages in environment at C:\Users\llamper\AppData\Local\Continuum\Anaconda3\envs\qcodes-minimal:
#
bleach                    1.5.0                    py36_0
certifi                   2016.2.28                py36_0
colorama                  0.3.9                    py36_0
cycler                    0.10.0                   py36_0
decorator                 4.1.2                    py36_0
entrypoints               0.2.3                    py36_0
h5py                      2.7.0               np113py36_0
hdf5                      1.8.15.1                 vc14_4  [vc14]
html5lib                  0.9999999                py36_0
hypothesis                3.23.0                   py36_0    conda-forge
icu                       57.1                     vc14_0  [vc14]
ipykernel                 4.6.1                    py36_0
ipython                   6.1.0                    py36_0
ipython_genutils          0.2.0                    py36_0
ipywidgets                6.0.0                    py36_0
jedi                      0.10.2                   py36_2
jinja2                    2.9.6                    py36_0
jpeg                      9b                       vc14_0  [vc14]
jsonschema                2.6.0                    py36_0
jupyter                   1.0.0                    py36_3
jupyter_client            5.1.0                    py36_0
jupyter_console           5.2.0                    py36_0
jupyter_core              4.3.0                    py36_0
libpng                    1.6.30                   vc14_1  [vc14]
markupsafe                1.0                      py36_0
matplotlib                2.0.2               np113py36_0
mistune                   0.7.4                    py36_0
mkl                       2017.0.3                      0
nbconvert                 5.2.1                    py36_0
nbformat                  4.4.0                    py36_0
notebook                  5.0.0                    py36_0
numpy                     1.13.1                   py36_0
openssl                   1.0.2l                   vc14_0  [vc14]
pandocfilters             1.4.2                    py36_0
path.py                   10.3.1                   py36_0
pickleshare               0.7.4                    py36_0
pip                       9.0.1                    py36_1
prompt_toolkit            1.0.15                   py36_0
py                        1.4.34                   py36_0
pygments                  2.2.0                    py36_0
pyparsing                 2.2.0                    py36_0
pyqt                      5.6.0                    py36_2
pyqtgraph                 0.10.0                   py36_0
pytest                    3.2.1                    py36_0
pytest-runner             2.11.1                   py36_0
python                    3.6.2                         0
python-dateutil           2.6.1                    py36_0
pytz                      2017.2                   py36_0
pyvisa                    1.8                      py36_0    conda-forge
pyzmq                     16.0.2                   py36_0
qcodes                    0.1.7                     <pip>
qt                        5.6.2                    vc14_6  [vc14]
qtconsole                 4.3.1                    py36_0
qtpy                      1.3.1                    py36_0
setuptools                36.4.0                   py36_1
simplegeneric             0.8.1                    py36_1
sip                       4.18                     py36_0
six                       1.10.0                   py36_0
testpath                  0.3.1                    py36_0
tk                        8.5.18                   vc14_0  [vc14]
tornado                   4.5.2                    py36_0
traitlets                 4.3.2                    py36_0
typing                    3.6.2                    py36_0
vs2015_runtime            14.0.25420                    0
wcwidth                   0.1.7                    py36_0
wheel                     0.29.0                   py36_0
widgetsnbextension        3.0.2                    py36_0
wincertstore              0.2                      py36_0
zlib                      1.2.11                   vc14_0  [vc14]`

jenshnielsen, 2017-09-26:
can you try

```
from pyqtgraph.python2_3 import basestring
print(isinstance(b'b', C:\\Users\\llamper\\AppData\\Local\\Continuum\\Anaconda3\\envs\\Python_3_5_ENV\\lib\\site-packages\\pyqtgraph\\graphicsItems\\PlotCurveItem.py:131: RuntimeWarning: All-NaN axis encountered\r\n))
```
and let me know the result

by the look of your traceback the function picks the wrong branch in cprint.py

jenshnielsen, 2017-09-26:
There is something odd on your system it should work in jupyter and does for me

lampertl, 2017-09-26:
No, I am saying it does work in jupyter but not from cmd.

lampertl, 2017-09-26:
The reason I am worried about running from jupyter, is that if I have a hybrid control, I will issue commands to the system via cmd and live plotting could have an issue. I'm starting with jupyter so there won't be any immediate issues. 

Thank you very much! I will continue from here and stay within jupyter at least until my scripts are running measurements.

jenshnielsen, 2017-09-26:
There should be no difference between running it from the jupyter notebook or otherwise. By cmd I assume that you mean Ipython or regular python launch from a cmd window as cmd in it self is a windows batch script interpreter and not a python one

jenshnielsen, 2017-09-26:
Ok I can reproduce it now. I misunderstood your original description. I thought you copied it to an interactive session. If I put it in a script and run the script I can reproduce it

lampertl, 2017-09-26:
Hmm, this is strange then. It runs well on jupyter and always throws the threading exception when I use anaconda prompt (python launch from cmd window).

lampertl, 2017-09-26:
OK, at least this means I am running something similar to your setup. Thank you for looking into this. I will keep progressing from here and run within jupyter.

jenshnielsen, 2017-09-26:
It's a bug in pyqtgraph and specifically the remote process plotting that we are using. This is known to be flaky and we are planning to replace it as soon as possible. 

The warning can be fixed by inserting 
```
if isinstance(arg, bytes):
    arg = arg.decode()
```
Just before the line 76 in cprint.py but that does not really fix the issue

lampertl, 2017-09-26:
Oh interesting! I'll add this into cpring.py as a band-aid for now.

lampertl, 2017-09-26:
Yes, noted. I want to avoid the warning for now if running through anaconda cmd. I'll still keep working with jupyter. Do you normally run all of your qcode work through jupyter?

jenshnielsen, 2017-09-26:
Most of our users tend to use spyder https://pythonhosted.org/spyder/ at the moment

jenshnielsen, 2017-09-26:
Spyder doesn't have this issue either

lampertl, 2017-09-26:
OK, do you recommend jupyter or spyder? I'm new to qcodes therefore I don't know of the pros/cons.

jenshnielsen, 2017-09-26:
If you are familiar with Matlab or other ide like envs then something like spyder will feel very familiar. Here you will probably write code in scripts and execute it interactively. 

The notebook allows you to write text mixed with code as a document so it's really a different style.

We have found that the ide interface is better for experimental code

lampertl, 2017-09-26:
Ah OK, that makes sense and I am familiar with ide like envs therefore I will give spyder a try. Thanks!

I am going to close the issue.

jenshnielsen, 2017-09-27:
I don't think we should close the issue as it's not really resolved we merely found a work around