microsoft/Qcodes #736: unhandy naming of parameter names in SR830 driver (and possibly others)

sohailc, 2017-09-14:
I would like to comment on a practice I have noticed in a number of places in the qcodes code base I have some comments on. For instance, the SR830 driver has parameters "chN_ratio", "chN_display", "chN_databuffer", where N is in [1, 2]. 

I do not understand why do we not have ch_ratio(N, ....) etc? What happens if in my experiment I want
to switch from channel 1 to 2? Change all numbers manually? Or what happens if I want to access channels one and two? Copy and paste code literally, instead of having a looping construct? Instead of the channel number being part of the parameter name, this should be an argument with which we call the parameter

This happens not only in this driver but in a number of them, and I do not think we should work this way.  What do you guys think?

jenshnielsen, 2017-09-14:
Thats why we added the channels feature but not all drivers have been converted at this stage. 
Compare for instance https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument_drivers/Keysight/Keysight_3
3500B.py to the new version at https://github.com/QCoDeS/Qcodes/blob/master/qcodes/instrument_drivers/Keysight/Keysight_33500B_channels.py

jenshnielsen, 2017-09-14:
In the old driver you would have to do 

```
key.ch1_function_type
```
and so on.
In the new version you can access all parameters on a channel such as
```
key.channels[0].function_type
```
and indeed do slicing too

sohailc, 2017-09-14:
Hi Jens, thanks for your reply. Good to see this is being addressed.

sohailc, 2017-09-14:
Is it ok if I make a similar change in the SR830 driver as well? I will send a PR in a few days.

jenshnielsen, 2017-09-14:
The workaround for old drivers is to use `getattr` such as 
```
getattr(key, 'ch{}_function_type'.format(i))
```
but that is obviously clunky.

sohailc, 2017-09-14:
Yeah I figured that work around, but if this needs to happen in a number of places in the code, this becomes *very* hard to read.

jenshnielsen, 2017-09-14:
If you want to try converting the SR830 thats great. You can use the keysight drivers as examples. We try to keep the old one around deprecated fro a while

sohailc, 2017-09-14:
Cool :-)

WilliamHPNielsen, 2017-09-15:
I guess we can close this issue. We can open a new one about how all drivers should be channelised and updated to the new parameter paradigm (no ManualParameters etc.).

closed 2017-09-15 by WilliamHPNielsen