microsoft/Qcodes #741: Deprecate/remove instrument.base.add_function ?

AdriaanRol, 2017-09-15:
Possibly related to #345 . 

Similar to the add_parameter, the instrument has a method `add_function`. To me the use case of this is very unclear and IMO opinion using regular python methods is preferred over using the add_function method to add a function as a method to an instrument. 

Advantages of using regular methods are
- more readable in the code/simpler 
- docstring generated are correct

As such I would propose to deprecate the use of the add_function method. 

@jenshnielsen @WilliamHPNielsen @sohailc 
(also now that Giulio left, who are all the remaining maintainers?)

WilliamHPNielsen, 2017-09-15:
@AdriaanRol, the hard, hard core team is the three of us whom you mentioned and @Dominik-Vogel.

WilliamHPNielsen, 2017-09-20:
I forgot to say: I agree with you. `add_function` is confusing clutter.

From the docs:
```
    This class is meant for simple cases, principally things that
    map to simple commands like '\*RST' (reset) or those with just a few
    arguments.
    It requires a fixed argument count, and positional args
    only. If your case is more complicated, you're probably better off
    simply making a new method in your Instrument subclass definition.
```