microsoft/Qcodes #745: Naming conventions to increase usability and lower entrance hurdle to QCoDes

Dominik-Vogel, 2017-09-18:
As a newcomer to QCodes I had some difficulties to get started with some simple instruments. One difficulty was owed to non-stringent naming of quantities. As an example consider the line
```
loop = qc.Loop(dac.ch03.v.sweep(0, 5, 0.1), delay=0.001).each(dmm.volt)
```
Here we loop over a channel of the QDac and read values with the Keysight34465A. I find it quite confusing that we are once using "volt" and once "v" and in other places "voltage".
I suggest we make a naming convention for commonly used properties, so that the drivers form a more integral code base.

jenshnielsen, 2017-09-18:
:+1: I think part of the problem is that we really need a good way to provide aliases for parameters.

Some users prefer short names as v over more verbose ones such as voltage, while other users finds it convenient to be able to address parameters by their physical connection to the sample i.e. plunger_v or some such thing. It would be useful if they were the same (taking snapshotting and so on into account so we dont get duplicated metedata)

Dominik-Vogel, 2017-09-18:
I totally agree.
I just noticed another point we can add to the list of driver conventions:
Naming the module:
instrument_drivers/<Manufacturer>/<Model> 
vs
instrument_drivers/<Manufacturer>/<Manufacturer>_<Model>