microsoft/Qcodes #754: ManualParameter calls validate twice

peendebak, 2017-09-21:
The qcodes `ManualParameter` calls validate twice for a single set.
``` python
import qcodes
from qcodes.utils.validators import Validator

class MyValidator(Validator):

    def __init__(self, min_value=-float("inf"), max_value=float("inf")):
        pass

    def validate(self, value, context=''):
        print('validate!')

    is_numeric=True

p = qcodes.ManualParameter('p', initial_value=0, vals=MyValidator())
p.set(10)
```
The output is:
```
validate!
validate!
```
The validator should only be called once.

@WilliamHPNielsen @jenshnielsen

WilliamHPNielsen, 2017-09-21:
I guess this is fixed by #651, which should be merged soon.

WilliamHPNielsen, 2017-10-17:
Hmm... it is not. I'll look in to this.

PR #794 (2017-10-19): fix double validation when setting parameters
Fixes #754 .

Changes proposed in this pull request:
- Don't double validate when setting

@jenshnielsen
closed 2017-10-19 by jenshnielsen