microsoft/Qcodes #757: Subplots with MatPlot

gzheng29, 2017-09-22:
MatPlot seems to have a subplot feature, but it's not obvious to use. Perhaps there could be a short tutorial on how to plot multiple heatmaps with certain figure sizes in a single figure.

nulinspiratie, 2017-09-25:
I added the latest changes to how to use the subplots, so i can add some info in a tutorial. @WilliamHPNielsen Should this go in the main QCoDeS tutorial, or in something separate?

WilliamHPNielsen, 2017-09-25:
@nulinspiratie, that's really nice of you! I think the `Tutorial.ipynb` should be left short so as to let newcomers get started quickly. You could start a new notebook called something like `Comprehensive Plotting How-To`. Then we'll (eventually) put all the advanced/non-standard plots in there.

closed 2017-09-28 by jenshnielsen

gzheng29, 2017-09-29:
Awesome guys, thanks a bunch!

PR #762 (2017-09-28): feat: add MatPlot guide
Fixes #757 

Changes proposed in this pull request:
- Adds comprehensive guide on plotting with MatPlot

This is basically all the tools I use in MatPlot, if anyone has additions please add it.
Also, it doesn't have anything yet on QTPlot, so could someone who uses that please add documentation?

@gzheng29 @WilliamHPNielsen