microsoft/Qcodes #765: Multiples validator only works for Ints

AdriaanRol, 2017-09-26:
The `Multiples(Ints)` validator checks if a give value is a multiple of some other value (the `divisor`). 

This validator only works for `Ints` and there is no reason for this. 

A perfectly valid example would be requiring for values to be multiples of 20ns. 

The current value would then be valid `40e-9` as this would return 0.0. Currently this will raise an error as 40e-9 (which is a multiple of 20e-9) raises an error as it is not an Int. I propose dropping the constraint of it having to be an Int.

jenshnielsen, 2017-09-27:
Thats sensible enough but we will nede to be careful about the implementation 
as floating point math cannot be assumed to be distributive. For example 

```python
In [1]: divisor = (1/3)*1e-10
In [2]: print(((1/3)*1e-9)%divisor)
3.333333333333331e-11
```

WilliamHPNielsen, 2017-11-03:
@AdriaanRol I'll put up an extra validator, `PermissiveMultiples`, for this on Monday.

PR #852 (2017-11-07): Validators/permissive multiples
Fixes #765

Changes proposed in this pull request:
- Add a validator for non-integer multiples of a base value
- Add a test (@AdriaanRol, perhaps you can think of more/better corner cases?)

The new validator, `PermissiveMultiples` allows integer multiples of a given base value to slip through. It allows negative multiples as well, and therefore by definition always 0. It is subject to rounding errors, so a user-specificied absolute precision is in principle required.

@jenshnielsen
closed 2017-11-07 by WilliamHPNielsen